MichaIng/DietPi #2226: DietPi-Software | Node.js: Running different versions concurrently

MichaIng, 2018-11-06:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=9&t=5209
- Different software might require different Node.js versions.
- Version can be changed via module `n`, but this will then be applied globally.
- Module `avn` seems to be a solution the run different versions concurrently, respectively apply different versions project/directory-wise: https://github.com/wbyoung/avn

borpin, 2018-11-11:
As a suggestion - for me I needed a specific version of node.js to run a module in node-red.

For that use case, providing a `dietpi-software` option to install n and then allow the user to select a version of node.js globally would have worked just fine.  As, on occasion, you install n anyway, this is not such a big change and still solves (some) of the use cases.

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+26

MichaIng, 2026-11-14:
Closing this. Our Node.js installer installs the latest LTS now, instead of latest stable. This should make the request mostly obsolete.

And with next build, our documentation will contain a section about how to use `n` to install custom Node versions: https://github.com/MichaIng/DietPi-Docs/blob/dev/docs/software/programming.md#nodejs

An info that is probably missing is that `n` replaces `/usr/local/bin/node`, which is needed for your Node-RED case, but may render the `node_modules` environment unusable, i.e. you might need to reinstall those manually (not using `dietpi-software`) so that all dependencies are resolved for the selected Node version. The same is true for any other Node app that may run with `/usr/local/bin/node`. We could use `n` in the first place, and use the versioned executable paths, but then e.g. `n 24` won't work without changing the path in the systemd unit manually.

This is all somewhat difficult to resolve, hence we (need to) assume applications support the latest LTS, which is the case for everything we offer in `dietpi-software`. We cannot assure it for every module, like we cannot assure it for every Python module and the system Python version. So this is then the responsibility of the admin.

closed 2026-11-14 by MichaIng