MichaIng/DietPi #2229: MotionEye on Tinkerboard

ibnbd, 2018-11-08:
I am not entirely sure why MotionEye doesn't support Tinkerboard. Tinkerboard is a good piece of hardware, can you make official support for MotionEye on it?

Thank You

MichaIng, 2018-11-08:
@ibnbd 
Thanks for your request.

On official releases there are only x86 and RPi releases available: https://github.com/Motion-Project/motion/releases

At least we can enable for amd64 then. Needs testing if we can get the RPi release work on other ARM devices or can be manually build there.

MichaIng, 2018-11-08:
Ah on Debian repo it is available for all architectures: https://packages.debian.org/stable/video/motion
On Jessie as well, but the version is outdated.

So generally enable for all devices, only on Jessie for RPi+amd64 user GitHub repo releases, otherwise rely on Debian repo. But needs testing thought all our supported devices and then enabled step by step at best, if succeeding. But yeah we can start with Tinkerboard.

@Fourdee 
What do you think?

Fourdee, 2018-11-09:
@MichaIng 

Yep 👍  we can enable for all under stretch>, but disable for Jessie.

ibnbd, 2018-11-09:
@MichaIng @Fourdee that will be amazing since `TinkerOS v2.0.7` is Debian stretch based.

Fourdee, 2018-11-09:
Test:
- 🈯️ ARMv7
- 🈯️ ARMv8

Completed. Available for all devices in v6.18.

closed 2018-11-09 by Fourdee