MichaIng/DietPi #2233: Use CFQ for HDDs

Generator, 2018-11-10:
---
name: Feature request
about: CFQ for better HDD performance

---

## About:
Dietpi sets the HDDs elevator to deadline on boot for all storage devices, for better performance HDD should use CFQ instead.

## Solution:

Create a udev rule to set HDD drives with CFQ and SDCARD/SSD devices with deadline

**Example:**
`/etc/udev/rules.d/60-ioschedulers.rules`
```
# set scheduler for non-rotating disks
ACTION=="add|change", KERNEL=="sd[a-z]|mmcblk[0-9]*|nvme[0-9]*", ATTR{queue/rotational}=="0", ATTR{queue/scheduler}="deadline"
# # set scheduler for rotating disks
ACTION=="add|change", KERNEL=="sd[a-z]", ATTR{queue/rotational}=="1", ATTR{queue/scheduler}="cfq"
```

$ cat /sys/block/sd*/queue/scheduler
noop deadline [cfq]
$ cat /sys/block/mmcblk*/queue/scheduler
noop [deadline] cfq

#### Additional context:

> The Completely Fair Queuing (CFQ) approaches the problem differently by allocating a timeslice and a number of allowed requests by queue depending on the priority of the process submitting them. It supports cgroup that allows to reserve some amount of I/O to a specific collection of processes. It is in particular useful for shared and cloud hosting: users who paid for some IOPS want to get their share whenever needed. Also, it idles at the end of synchronous I/O waiting for other nearby operations, taking over this feature from the anticipatory scheduler and bringing some enhancements. Both the anticipatory and the elevator schedulers were decommissioned from the Linux kernel replaced by the more advanced alternatives presented above.   

https://wiki.archlinux.org/index.php/Improving_performance#Input.2Foutput_schedulers

Fourdee, 2018-11-11:
@Generator 

Many thanks for the info 👍 

We'll review this when possible and assess if a roll-out is beneficial.

MichaIng, 2019-06-03:
@Generator 
I want to implement this with v6.25.

`deadline` is btw only for RPi and done by default `cmdline.txt` that comes with Raspbian. Indeed this does not make much sense rot rotating disks.

One question is:
- I am not 100% sure if all kernels that we ship supports rotational/non-rotational detection, although sure enough. Just read on old docs that `/sys/block/*/queue/rotational` does/did not always exist or show correct values.
- Perhaps we should allow to set custom schedulers for each drive via `dietpi-drive_manager`? Do you know if it is possible to add additional udev rules for specific drives that override the ones you suggested above? Although I could just test this myself quickly 😉.

Generator, 2019-06-03:
The only other way that i know is with `lsblk`

```
$ lsblk -d -o name,rota
NAME    ROTA
sda        1
mmcblk0    0
```
See [5 Commands to Know If a Disk is SSD or HDD in Linux](https://techglimpse.com/disk-ssd-hdd-linux-commands/)

MichaIng, 2019-06-03:
@Generator 
As above, we can anyway not ship fixed udev rules since the supported schedulers are different.
On VM Stretch (kernel 4.9) it's as one finds it on most online docs and guides about this topic, CFQ set by default, as we do not touch it:
```
root@VM-Stretch:~# cat /sys/block/sda/queue/scheduler
noop deadline [cfq]
```
On VM Buster (kernel 4.19) as well mq-deadline (like RPi kernel 4.19) is available, but no CFQ or BFQ at all:
```
root@VM-Buster:~# cat /sys/block/sda/queue/scheduler
[mq-deadline] none
```
Quick search didn't show much about kyber, besides it is developed by Facebook 🤔.

MichaIng, 2019-06-09:
PR up: https://github.com/MichaIng/DietPi/pull/2902

closed 2019-07-03 by MichaIng