MichaIng/DietPi #2237: Pi-hole doesn't install properly

Kreeblah, 2018-11-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| v6.17.12
- Distro version	| stretch
- Kernel version	| Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux
- SBC device		| Virtual Machine (x86_64)
- Power supply used	| N/A (VMWare Fusion 11.0.1)
- SDcard used		| N/A (VMWare Fusion 11.0.1)

#### Additional Information (if applicable):
- Software title	| Pi-hole
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bugreport ID: 80e3b5f9-5236-4dbf-adf2-c04397740817

#### Steps to reproduce:

1. Install a fresh copy of DiePi from the VMWare image and let it upgrade to the current release and then reboot.
2. Set a static IP address in the DietPi network settings
3. Install Pi-hole from the DietPi optimized software list

#### Expected behaviour:

- Pi-hole should install properly

#### Actual behaviour:

- Going to /pihole on the IP address (as noted on https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p174) returns a 404

#### Extra details:

- There were a ton of getcwd() and file not found errors during the installation
- dietpi-software displays an error saying that /etc/init.d/pihole-FTL doesn't exist or can't be written to as the current user (which was root)

Fourdee, 2018-11-12:
@Kreeblah 

Many thanks for the report 👍 

> here were a ton of getcwd() and file not found errors during the installation

Interesting, ok.
I'll try to replicate

Testing:
- 🈯️ ARMv7 beta
- VM master

Fourdee, 2018-11-12:
Beta:
[image]

Fourdee, 2018-11-12:
Using a fresh image, that wasn't used for previous dev testing:
🈯️ Fresh VB image with 2.1GB RAM
🈴 Fresh VB image with 1GB RAM

Fourdee, 2018-11-12:
```
[ INFO ] DietPi-Software | Entered scripts working directory: /tmp/DietPi-Software
[  OK  ] DietPi-Software | Initialized database
[ .... ] DietPi-Software | Reading database, please wait...

G_EXIT for DietPi-Software
PWD=/tmp/DietPi-Software
[  OK  ] DietPi-Software | Reading database completed
[ INFO ] DietPi-Software | Navigated to /tmp
```

Fourdee, 2018-11-12:
Before sync fix:
- Notice the two G_EXIT calls.
```
root@DietPi:~# G_DEBUG=1 dietpi-software install 93
G_DIETPI-NOTIFY CALLED
[  OK  ] DietPi-Software | Root access verified.
G_DIETPI-NOTIFY CALLED
[  OK  ] DietPi-Software | RootFS R/W access verified.

G_DIETPI-NOTIFY CALLED

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Running G_INIT()

G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Entered scripts working directory: /tmp/DietPi-Software
G_DIETPI-NOTIFY CALLED
[ .... ] DietPi-Software | Initializing databaseDietPi-Software | 32396 > /tmp/dietpi-process.pid (EXIT)
[.     ] G_DIETPI-NOTIFY CALLED
/tmp/dietpi-process.pid PID KILL (EXIT)
CLEARING process animation
[  OK  ] DietPi-Software | Initialized database
DEBUG Read_InstallFileList 0
G_DIETPI-NOTIFY CALLED
[ .... ] DietPi-Software | Reading databaseDietPi-Software | 32402 > /tmp/dietpi-process.pid (EXIT)
DEBUG Read_InstallFileList 1
DEBUG Read_InstallFileList 2
DEBUG Read_InstallFileList 3
DEBUG Read_InstallFileList 4
G_DIETPI-NOTIFY CALLED
/tmp/dietpi-process.pid PID KILL (EXIT)
DietPi-Software Running G_EXIT()
G_DIETPI-NOTIFY CALLED
/tmp/dietpi-process.pid PID KILL (EXIT)
CLEARING process animation
[  OK  ] DietPi-Software | Reading database
DEBUG Read_InstallFileList 5
DietPi-Software Running G_EXIT()
G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Navigated to /tmp
G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Removed scripts working directory: /tmp/DietPi-Software
root@DietPi:~# CLEARING process animation
[ INFO ] DietPi-Software | Navigated to /tmp
G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Failed to removed scripts working directory: /tmp/DietPi-Software
```

After sync fix:
```
root@DietPi:~# G_DEBUG=1 dietpi-software install 93
G_DIETPI-NOTIFY CALLED
[  OK  ] DietPi-Software | Root access verified.
G_DIETPI-NOTIFY CALLED
[  OK  ] DietPi-Software | RootFS R/W access verified.

G_DIETPI-NOTIFY CALLED

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Running G_INIT()

G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Entered scripts working directory: /tmp/DietPi-Software
G_DIETPI-NOTIFY CALLED
[ .... ] DietPi-Software | Initializing databaseDietPi-Software | 32281 > /tmp/dietpi-process.pid (EXIT)
[.     ] G_DIETPI-NOTIFY CALLED
/tmp/dietpi-process.pid PID KILL (EXIT)
CLEARING process animation
[  OK  ] DietPi-Software | Initialized database
DEBUG Read_InstallFileList 0
G_DIETPI-NOTIFY CALLED
[ .... ] DietPi-Software | Reading databaseDietPi-Software | 32288 > /tmp/dietpi-process.pid (EXIT)
[.     ] DEBUG Read_InstallFileList 1
DEBUG Read_InstallFileList 2
DEBUG Read_InstallFileList 3
DEBUG Read_InstallFileList 4
G_DIETPI-NOTIFY CALLED
/tmp/dietpi-process.pid PID KILL (EXIT)
CLEARING process animation
[  OK  ] DietPi-Software | Reading database
DEBUG Read_InstallFileList 5
DietPi-Software Running G_EXIT()
G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Navigated to /tmp
G_DIETPI-NOTIFY CALLED
[ INFO ] DietPi-Software | Removed scripts working directory: /tmp/DietPi-Software
```

Fourdee, 2018-11-13:
Needs more testing.
- Use PID under a GLOBAL, exclude filesystem issue

Fourdee, 2018-11-13:
@MichaIng 

> This was just a test, right? 

Yep, all above was debug testing, trying to find cause of the unexpected ```G_EXIT``` call.

> We should clean /tmp before mounting tmpfs there during PREP or whenever done within our scripts

yep 👍 

> And just to be sure, this issue does not only occur on first run install then, right?

Occurs after 1st run installation, and during boot (briefly on VM):
[image]

Fourdee, 2018-11-13:
@MichaIng 

Fixed the boot issue, use the binary instead of variable ```$PWD``` to get current directory:
```
if [[ $(pwd) == /tmp/$G_PROGRAM_NAME* ]]; then
```

Fourdee, 2018-11-13:
Regarding traps:

> http://redsymbol.net/articles/bash-exit-traps/
If some error causes the script to exit prematurely, the scratch directory and its contents don't get deleted. This is a resource leak, and may have security implications too.

This is defiantly the cause of the ```dietpi-software``` issues. Disabling this resolves.
```
G_DIETPI-NOTIFY -2 'Reading database'
```

Fourdee, 2018-11-13:
@MichaIng 

Think i've fixed it:
- Using SIGTERM kill
- Tested ```G_DEBUG=1 dietpi-software install 93``` as before, no errors!
```
kill -9 $(</tmp/dietpi-process.pid) &> /dev/null
```

MichaIng, 2018-11-13:
> This is definitely the cause of the dietpi-software issues. Disabling this resolves.

I don't get the connection yet. The AMI instance in the guides example is our animation process? If the `dietpi-software` exit trap does not terminate it, it will run forever. Jep makes sense, for this reason currently the PS1 prompt command terminates the animation as last resort. However it makes sense to do this within the exit trap as well. PS1 prompt command will then only do this, if some G_* command from terminal calls animation and is cancelled or fails.

About `$PWD`:
- E.g. if G_INIT creates the directory and `cd` into it and G_EXIT is called very fast afterwards, leading to `$PWD` being not yet updated accordingly, it will lead to `cd` back into `/tmp` is not done and working dir removal leads to getcwd error. Makes sense.
- But the working dir then definitely should have been created, otherwise `cd` should not even have been possible. So `async` then is not the problem, just the outdated `$PWD` variable? Then `$(pwd)` indeed is a good solution to assure correct current dir is checked.

Perhaps it is easiest/fastest to simply always `cd /tmp` before removing working dir and skip the test completely?

But now about the actual debug log:
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Running G_INIT()

[ INFO ] DietPi-Software | Entered scripts working directory: /tmp/DietPi-Software
[  OK  ] DietPi-Software | Initialized database
[ .... ] DietPi-Software | Reading database, please wait.../tmp/DietPi-Software
total 8.0K
drwxr-xr-x 2 root root 4.0K Nov 12 18:18 .
drwxrwxrwt 8 root root 4.0K Nov 12 18:18 ..
[ INFO ] DietPi-Software | Navigated to /tmp
[ INFO ] DietPi-Software | Removed scripts working directory: /tmp/DietPi-Software
[  OK  ] DietPi-Software | Reading database completed
```
- Working dir is created and surely synced, since:
  1. `[[ -d /tmp/DietPi-Software ]]` returned true, since:  `[ INFO ] DietPi-Software | Removed scripts working directory: /tmp/DietPi-Software`
  2. `[[ $PWD == /tmp/$G_PROGRAM_NAME* ]]` returned true as well, so `$PWD` obviously was correct, since: `[ INFO ] DietPi-Software | Navigated to /tmp`

Thinking now about this, I believe it is due to:
`/DietPi/dietpi/func/dietpi-set_dphys-swapfile` remounting `/tmp`, which leads to content and cwd loss.
This is called during `preboot` 1st run setup, after which the error appeared and during Pi-hole install, after which error was reported. 🈯️ Makes totally sense!
- We need to:
  1. backup `/tmp` content
  2. `cd` out of `/tmp`
  3. `umount /tmp`
  4. Remove obsolete possibly existent `/tmp` content
  5. re-`mount /tmp`, which will automatically mount according to fstab entry, AFAIR (as far as I read 😉) but need test
  6. recover content
  7. `cd /tmp/$G_PROGRAM_NAME`, which should be the swapfile scripts working dir.

- [ ] Verify that parent script pwd is still correct, even that /tmp was remounted in between.

But how does `pwd` now fix the issue? Perhaps it also corrects the current dir, if non-existent, re-creates it or sets it to parent dir or $HOME or such?

Last question is still why the `dietpi-software` EXIT trap was actually called during database reading above. /tmp remount and wrong $PWD cannot be the issue since on 1st run setup, dietpi-software is definitely called long after preboot resets /tmp mount and database reading is done before Pi-hole install code remounts it.
Since the script goes on, no EXIT call was done, obviously.
- [ ] So still need to check about EXIT traps being passed to child background processes.

Puhh looks to me that all of these issues are not related to each other. The real issue, which is the cause for the TO and your replicated 1st boot/install error is definitely swapfile creation remounting /tmp, leading to content being cleared.

Fourdee, 2018-11-13:
@MichaIng 

> Thinking now about this, I believe it is due to:
/DietPi/dietpi/func/dietpi-set_dphys-swapfile remounting /tmp, which leads to content and cwd loss.
This is called during preboot 1st run setup, after which the error appeared and during Pi-hole install, after which error was reported. 🈯️ Makes totally sense!

The boot issue occurs at all times, even after 1st run setup.

MichaIng, 2018-11-13:
@Fourdee 
Also, also found swap file creation, including `mount -o remount,size=753M tmpfs /tmp` did not clear /tmp dir. Current $PWD and even contained files are reserved. Seems the content backup is somehow done automatically 👍.

Fourdee, 2018-11-13:
@MichaIng 

> Tested exit traps up and down with/without disown, immediate and after a while terminating, from inside and outside the background job, within/outside { ... } &> ...

```
[ INFO ] DietPi-Software | Entered scripts working directory: /tmp/DietPi-Software
[.     ] /DietPi/dietpi/func/dietpi-globals: line 270:  2455 Terminated              Print_Process_Animation
[  OK  ] DietPi-Software | Initialized database
[ INFO ] DietPi-Software | Navigated to /tmp
[  OK  ] DietPi-Software | Reading database completed
[ INFO ] DietPi-Software | Removed scripts working directory: /tmp/DietPi-Software
/DietPi/dietpi/dietpi-software: line 1:  2461 Terminated              Print_Process_Animation
```

MichaIng, 2018-11-13:
@Fourdee 
> Would indicate the process is not terminated fast enough using SIGTERM

But why this is an issue? Even if the kill command does it's job in the background, while script goes on, this would at worst lead to a parallel animation, until kill has finished. But this should not lead to any folder deletion or affect cwd?
_________
Quick validation about tmpfs remount: http://man7.org/linux/man-pages/man5/tmpfs.5.html
>        *  During a remount operation (mount -o remount), the filesystem size
>           can be changed (without losing the existing contents of the
>           filesystem).

👍

Fourdee, 2018-11-13:
@MichaIng 

> But why this is an issue? Even if the kill command does it's job in the background, while script goes on, this would at worst lead to a parallel animation, until kill has finished. But this should not lead to any folder deletion or affect cwd?

Unsure at moment. Maybe SIGTERM is allowing a process/memory leak to occur in ```Print_Process_Animation``` when terminated.

> Quick validation about tmpfs remount: 

Nice 👍

Fourdee, 2018-11-13:
Intresting, this works:
```
[[ -w /tmp/dietpi-process.pid ]] && echo -ne "\r$bracket_l${aprocess_string[i]}$bracket_r " || tput ed && return
```

Then simply ```rm /tmp/dietpi-process.pid``` in process clean.

So issue is leakage when terminating ```Print_Process_Animation```?

Fourdee, 2018-11-13:
@MichaIng 

If you need VNC access for the boot issue, let me know i'll set it up?

MichaIng, 2018-11-13:
@Fourdee 
But the above is already live code? Only `tput ed` added, but what is the influence of this?
Ah, not required in `Clean_Process_Animation` then. But what I don't like about it:
- If `tput` fails for another reason, animation does not stop. Okay but simple solution is `|| { tput ed; return; }`
- If for some unexpected reason the file is removed after other output was done as well, `tpud ed` would possibly remove this other output and/or we would like to still see the last processing message, if it was terminated unexpectedly? I like to have this at defined positions, when G_DIETPI-NOTIFY wants to overprint, or script exit and PS1 prompt want to terminate and clean output.

I still not get it 🤔. Made more research and more testing and still unable to find any way to make a child termination call the parent exit trap... And even if something goes wrong with the PID, e.g. somehow parent PID is saved to PID file, then the parent script would exit as well, which it does not...

So even that we have some solutions with `cd /tmp` and `$(pwd)` that seem to solve the issue, I would like to understand how dietpi-software exit trap can be called, when child background process is terminated and without dietpi-software actually exiting. So it receives an EXIT signal but goes on working?? I have headache now 🤣.

Fourdee, 2018-11-13:
@MichaIng 

TightVNC works well. Setting it up now.

Fourdee, 2018-11-13:
@MichaIng 

82.7.94.230 same pw as webserver.

MichaIng, 2018-11-14:
Okay long testing session:
- `/DietPi/dietpi/preboot` is actually fine. It looks like it calls exit trap before it's done, but it's not. All remaining tasks are send to background and exit traps obviously do not wait for background children to finish, which is good to know.
- `cd /tmp` will be done during exit trap without any `$PWD`/`$(pwd)` or [[ -d /tmp/$G_PROGRAM_NAME ]] checks. Prevents issues due to possible `async` and/or `$PWD` update delay and if `/tmp/$G_PROGRAM_NAME` was removed for another reason.
- `dietpi-software install 93` still needs investigation. However process animation will be killed via `kill -9` (SIGKILL vs SIGTERM), just in case and it's faster.

Fourdee, 2018-11-14:
@MichaIng 

Thanks Micha, indeed a very long debugging session 😄  Thanks for your help 👍 

Some notes my end:

> dietpi-software install 93 still needs investigation. However process animation will be killed via kill -9 (SIGKILL vs SIGTERM), just in case and it's faster.

Removal of kill command and allowing bg process to terminate on its own when ```.pid``` file is removed, also worked.

Any Pipe with preboot script and multiple + threads in the script, was triggering the getcwd errors during boot.

MichaIng, 2018-11-14:
> Removal of kill command and allowing bg process to terminate on its own when .pid file is removed, also worked.

The only issue with this is the max 0.15 seconds delay between removal and termination:
- Animation 1 runs
- Animation 1 PID file is removed, sleep time until next loop + check is still 0.1 seconds
- 0.05 seconds afterwards Animation 2 is requested and started, since PID file does not exist
- Animation 1 loop reaches check and goes on, since still a PID file exists.
- Both animations run concurrently and theoretically on very fast systems those can stack up.

This only works, if the animation process checks content of PID file, verifying that it's still his own PID `[[ $(</tmp/dietpi-process.pid) == $BASHPID ]]`. But decreases performance 🤔.
For my impression active process termination before removing PID file (thus allowing new animations) is the better deal then.

> Any Pipe with preboot script and multiple + threads in the script, was triggering the getcwd errors during boot.

Still not sure, where exactly the getcwd error came from, but they did not appear after forced `cd /tmp` any more, right? At least the multiple bg processes were not related.

Fourdee, 2018-11-15:
@MichaIng 

> Still not sure, where exactly the getcwd error came from, but they did not appear after forced cd /tmp any more, right? At least the multiple bg processes were not related.

Still occurring on my tests. 

Believe you were right with a ```/tmp``` mount issue as following does not create log:
```
ExecStart=/bin/bash -c '/DietPi/dietpi/preboot &>> /tmp/dietpi-preboot.log'
```

🈯️ However, this does:
```
ExecStart=/bin/bash -c '/DietPi/dietpi/preboot &>> /root/dietpi-preboot.log'
```

Fourdee, 2018-11-15:
@MichaIng 

Ok, still unsure of cause. Although:
- 🈯️ disabling threading in ```preboot``` resolves issue.

So for now, I believe we should roll out a workaround fix to disable threading, until more time is available to debug this further?

---

Also tried:
```G_THREAD_START```, instead of ```&```, same errors in output.

MichaIng, 2018-11-15:
@Fourdee 
```
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
[  OK  ] Root access verified.
[  OK  ] Root access verified.
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
```
- I believe this occurs within the cpu and led background jobs (both two times and perfectly around their root user check). `cd /tmp` seemed to be done successfully (already prior to change), but the background jobs then seem to fail getting current directory.
- Perhaps, when initiated, they expect to be within `/tmp/DietPi-PreBoot`, as this was cwd when being initiated. But immediately parent script changes to `/tmp` and removes cwd of background jobs.

Does it solve to actively do `cd /tmp` before initiating background jobs?

Fourdee, 2018-11-15:
@MichaIng 

> Does it solve to actively do cd /tmp before initiating background jobs?

Nope, tried ```cd /tmp; command &```, still occurs

MichaIng, 2018-11-15:
Best would be to let the trap wait for those to finish: https://stackoverflow.com/a/356154
- Store PIDs in variable and use `wait`.
- Or hmm, is the reason simply the active `exit` call at end of `preboot`? Should be not required, also we do not pass any exit code.

So test schedule:
- [ ] Remove `exit` command from end of `preboot`
- [ ] Store initiated background process PIDs in variable and use `wait` at end of script (before `exit`).
- [ ] Active `cd /tmp` before initiating background jobs, although this means we need to keep that in mind, in case add tmp file creation later.

Fourdee, 2018-11-15:
Lowering priority level as both Stretch VM images have been updated with the fix:
- https://dietpi.com/downloads/images/DietPi_VMWare-x86_64-Stretch.7z
- https://dietpi.com/downloads/images/DietPi_VirtualBox-x86_64-Stretch.7z

MichaIng, 2018-11-20:
Okay went on with testing:
- Made `set_cpu` and `set_led` being bg jobs again and added `G_DEBUG=1` to top of preboot.
- Suddenly I also received the `getcwd` errors, but not consistently. One time they showed up for one of set_cpu/led only, one time it showed up for both.
- It always throws on `shell-init` (init of bg job) and on `chdir` (when G_INIT want's to `cd` into subscript working dir).

As mentioned above, after the bg jobs are initiated, preboot does not wait for them to finish with the EXIT trap: `cd /tmp` and `rm -R /tmp/DietPi-PreBoot`. Most properly the bg job init (+ `cd /tmp/$G_PROGRAM_NAME`) and the EXIT traps steps are too close together, breaking each others sometimes.

Fourdee, 2018-11-20:
@MichaIng 

🈯️ ```G_THREAD_*```
[image]

MichaIng, 2018-11-20:
@Fourdee 
Ah lol jep, totally forgot that we already have the function set for this 👍.

Jep should totally work with this. However thread internal output is missing then and still the question is, if there is really a boot time benefit with this. Needs to be tested on non-VM, I guess, where set_cpu/led really apply changes.

Fourdee, 2018-11-20:
Testing:
- ARMv8 + settings applied

Fourdee, 2018-11-24:
Marking as completed. Issue is now resolved by waiting for background threads to finish, before script exit.

closed 2018-11-24 by Fourdee