MichaIng/DietPi #2238: Docker install fails to startup after installation

mspieth376, 2018-11-13:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.17.12
- Distro version	| 9.6
- Kernel version	| Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux
- SBC device		| Virtual Machine (x86_64) also on RP3 and UX4
- Power supply used	| N/A
- SDcard used		| N/A

#### Additional Information (if applicable):
- Software title	| docker
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. Install Docker via dietpi-software
2. Docker will fail to start via systemd

#### Expected behaviour:

Docker should start

#### Actual behaviour:

service docker start
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.

#### Extra details:

The problem is the userdata location line when docker is started.

In /DietPi/dietpi/dietpi-software there is a line:

# Set container(s) locations in /lib/systemd/system/docker.service
sed -i "/ExecStart=\/usr\/bin\/dockerd/c\ExecStart=\/usr\/bin\/dockerd -g $G_FP_DIETPI_USERDATA\/docker-data -H fd:\/\/" /lib/systemd/system/docker.service

This line breaks docker startup by setting  /lib/systemd/system/docker.service to
ExecStart=/usr/bin/dockerd -g /mnt/dietpi_userdata/docker-data -H fd://

This should be
ExecStart=/usr/bin/dockerd -g /mnt/dietpi_userdata/docker-data -H unix://

MichaIng, 2018-11-14:
@mspieth376 
Thanks for report and fix. Will validate on Jessie and Buster systems as well, then merge accordingly for v6.18.

MichaIng, 2018-11-15:
Testing:
- [x] Stretch: fd:// does not work, unix:// does
- [x] Jessie: fd:// DOES work!, unix:// works as well
- [x] Buster: fd:// does not work, unix:// does

Interesting, official packaging repo service unit still contains fd://: https://github.com/docker/docker-ce-packaging/blob/master/deb/systemd/docker.service
- Höh, there are two different files on the repo: https://github.com/docker/docker-ce-packaging/blob/master/systemd/docker.service 🤔
However without touching the file, it's unix:// on Stretch + Buster, on Jessie still fd://

Generally I don't like editing systemd units directly. If really, better move it into a `docker.service.d/dietpi.conf` file, but:
- https://docs.docker.com/config/daemon/#configure-the-docker-daemon
- https://docs.docker.com/config/daemon/systemd/#runtime-directory-and-storage-driver
- Better solve it via config file. This way the service unit stays completely untouched and by this can be e.g. auto updated by APT.

Will open PR...

Fourdee, 2018-11-16:
Testing for running of Docker after install + reboot:
- 🈯️ VM
- 🈴 RPi, service hangs on start
```
root@DietPi:~# systemctl status docker -l
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; disabled; vendor preset:
enabled)
   Active: activating (start) since Fri 2018-11-16 16:51:21 GMT; 3min 32s ago
     Docs: https://docs.docker.com
 Main PID: 944 (dockerd)
      CPU: 340ms
   CGroup: /system.slice/docker.service
           └─944 /usr/bin/dockerd -H unix://

Nov 16 16:51:21 DietPi systemd[1]: Starting Docker Application Container Engine.
..
```

Fourdee, 2018-11-16:
Hmm, with RPi, ```systemctl start docker``` hangs during boot. 
- Does not occur every reboot. 
- Clears after 10-60 seconds
- Running docker images during the service hangs, works fine.

Downside is it prevents ```dietpi-services start``` from continuing, until Docker is finished loading.

Will set service type to simple, to prevent the above.

Fourdee, 2018-11-16:
Retest:
- 🈯️ VM
- 🈯️ RPi

@MichaIng 

Great work Micha on fix, works a charm, completed.

closed 2018-11-16 by Fourdee

MichaIng, 2018-11-16:
@Fourdee
Great non-x86 fix as well.

As long as we cannot find the reason for the hang, this is the best we can do.

Perhaps docker requires something else to start-up? Is there some log that gives an indication?
But actually funny since dietpi services starts it at end of boot, later than how systemd would do :thinking:.

So default APT install with systemd control should have the same issue then, perhaps we can find something about it in web, RPi + docker...

garret, 2018-11-20:
Hello everybody, I believe to be the main responsible to have found out about this bug (https://dietpi.com/phpbb/viewtopic.php?f=11&t=5227) but to me the workaround does not work. 
Do you know if there is a solution or if you want I can paste some logs if you tell me which commands to give. Sorry, I am very newbie to Docker.

MichaIng, 2018-12-10:
Lol, on official repo, `fd://` is again in service file, working well now on Stretch...
Indeed best to not touch service file directly, if it is changed back and forth 😄.