MichaIng/DietPi #2239: DietPi-Software | Drop SiCKRAGE, enable SickChill

MichaIng, 2018-11-13:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5163&p=15370#p15344

Looks like SiCKRAGE is simply dropped from FOSS development. As well install seems to be (again) broken. With SickChill we already found that old install method works, so can be recovered easily.

Fourdee, 2018-11-13:
@MichaIng 

What a mess.
I'am eager to remove Sickrage and put a ban on all forks of sickrage being implemented into DietPi. 

I do not believe we should be spending our time, wasted on chasing up program switch overs due to childish endeavours. All forks of Sickrage at at risk of this in the future.

We already offer Sonarr, which, in my view is vastly superior in every regard of performance, usability, stability.

> https://www.reddit.com/r/synology/comments/9om1iq/latest_sickrage_news_and_how_it_affects_you/e7v003q

So, I believe we should?
- Drop Sickrage
- Do not implement any fork of Sickrage, in any form.
- Continue to offer Sonarr, which is vastly superior, and, professional devs.

mrmajik45, 2018-11-13:
So give SiCKRAGE the L and kick it out, got it.

MichaIng, 2018-11-14:
@Fourdee 
Jep perhaps best, until we get more requests and the different forks become settled.

NB: If I understand right, SickChill is actually the direct follower of SickRage, just forced to be renamed. So we simply made the mistake to switch to "SiCKRAGE", which looks like the direct follower, but actually changed the way it is installed and such and obviously still changes much, which makes our install method constantly (re)fail.

But jep, okay to drop all forks for now. Instead as alternative to Sonarr, we can have a look at: https://github.com/pymedusa/medusa
- But at best we move that to v6.19 to not overload and delay v6.18 too much. Better focus on more testing and intense Beta phase maintenance.

Fourdee, 2018-11-15:
@MichaIng 

> But at best we move that to v6.19 to not overload and delay v6.18 too much. Better focus on more testing and intense Beta phase maintenance.

Yep agree. Will disable SickRage install in v6.18, then we can re-asses this in v6.19.

MichaIng, 2018-11-22:
Votes for Medusa: https://dietpi.com/phpbb/viewtopic.php?f=12&t=3024

mrmajik45, 2018-11-26:
I'll try and help code Medusa :P

JackDandy, 2018-11-30:
Perhaps try [SickGear](https://github.com/SickGear/SickGear) - with none of the rage/medusa code it also has none of their issues/dramas.

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+17

MichaIng, 2018-12-26:
Further reading: https://www.reddit.com/r/sickrage/comments/9myqx0/can_some_give_me_a_run_down_on_what_is_going_on/
- So indeed SickChill is the real prior SickRage.
- However Medusa is an older fork, so should be more stable/settled in terms of devs and dev.
- The additional feature are just from the side of a Medusa dev, but really, I believe this is the best way to go.

Fourdee, 2018-12-31:
Install test + download test with transmission + Medusa web update test + uninstall test:
- 🈯️ ARMv7
- 🈯️ ARMv8

@MichaIng 

Works a charm, nice one 👍 

Forum doc matches, just need to 
- change references to Medusa from Sickrage and images

closed 2019-01-28 by Fourdee