MichaIng/DietPi #2240: Hash Sum mismatch

rucknapucknavitz, 2018-11-14:
#### Details:
- Date           | Wed 14 Nov 03:02:55 GMT 2018
- Bug report     | N/A
- DietPi version | v6.17.0 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. Burn fresh image
2. Boot device

#### Expected behaviour:

- Boot to install

#### Actual behaviour:

- Error

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Get:1 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 https://archive.raspberrypi.org/debian stretch/main armhf Packages [181 kB]
Get:4 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [34.3 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Err:5 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
  Hash Sum mismatch
  Hashes of expected file:
   - Filesize:11660768 [weak]
   - SHA256:5ec93d86507956b45b41fa4a4cec711f65362cf513e01a5f0b9b590a9eb77a85
   - SHA1:4d6886eba974844f45900f1658331d3d37f4b962 [weak]
   - MD5Sum:386517fb7b2b843239d3b1229ea5d671 [weak]
  Hashes of received file:
   - SHA256:09eaa27e08c592b12eb18feb9cdf87aaa533872ef3c3fae946d3c4d1241cc41a
   - SHA1:62a382f289d99cb6238d060e696d2d4607120f72 [weak]
   - MD5Sum:dcfa3ce1c62fec4bb5c605e34f1b861e [weak]
   - Filesize:1966121 [weak]
  Last modification reported: Tue, 13 Nov 2018 04:32:57 +0000
  Release file created at: Tue, 13 Nov 2018 04:35:22 +0000
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages [56.9 kB]
Get:7 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Get:8 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages [1,360 B]
Fetched 2,375 kB in 4min 41s (8,450 B/s)
Reading package lists...
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages.xz  Hash Sum mismatch
   Hashes of expected file:
    - Filesize:11660768 [weak]
    - SHA256:5ec93d86507956b45b41fa4a4cec711f65362cf513e01a5f0b9b590a9eb77a85
    - SHA1:4d6886eba974844f45900f1658331d3d37f4b962 [weak]
    - MD5Sum:386517fb7b2b843239d3b1229ea5d671 [weak]
   Hashes of received file:
    - SHA256:09eaa27e08c592b12eb18feb9cdf87aaa533872ef3c3fae946d3c4d1241cc41a
    - SHA1:62a382f289d99cb6238d060e696d2d4607120f72 [weak]
    - MD5Sum:dcfa3ce1c62fec4bb5c605e34f1b861e [weak]
    - Filesize:1966121 [weak]
   Last modification reported: Tue, 13 Nov 2018 04:32:57 +0000
   Release file created at: Tue, 13 Nov 2018 04:35:22 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

rucknapucknavitz, 2018-11-14:
This seems to have been resolved. Perhaps an issue on the raspberry pi Raspbian side of things.

closed 2018-11-14 by rucknapucknavitz

tripflip, 2019-03-02:
same problem here, rpi3b, freshly burned img, and I'm getting the same errors