MichaIng/DietPi #2241: /DietPi/dietpi/dietpi-cleaner: line 171: Run_Cleaner: command not found

optio50, 2018-11-14:
when running cleaner I get

/DietPi/dietpi/dietpi-cleaner: line 171: Run_Cleaner: command not found

The test clean seems to go ok but not clean

Raspberry Pi Zero W Rev 1.1
Linux 4.14.80+ armv6l
DietPi_v6.17_RPi-ARMv6-Stretch

Fourdee, 2018-11-14:
@optio50 

Many thanks for the report 👍 , we'll try to replicate + resolve.

Fourdee, 2018-11-14:
@optio50 

Resolved in v6.18.

### Solution
- Update script with the fixed code
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/7abf046c47ca01469eac6935b5ce19a4067a38cb/dietpi/dietpi-cleaner -O /DietPi/dietpi/dietpi-cleaner
```

closed 2018-11-14 by Fourdee

MichaIng, 2018-11-14:
Whoopsie, that was my fault: https://github.com/Fourdee/DietPi/commit/f18f55e44a02fec2c82949b4f1a44dc7b57aea92#diff-97cafd2665ea39c9a89ac02d92dffcad

Note to self: Every touched file needs full testing of branch after merge: Missed that one it seems.

optio50, 2018-11-17:
Seems to work. Thanks :-)