MichaIng/DietPi #2242: wordpress not works on fresh install

msongz, 2018-11-14:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=12
G_GITBRANCH=master
G_GITOWNER=Fourdee`
- Distro version	| `stretch` or `9.4`
- Kernel version	| `Linux pi2 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`
- SBC device		| `RPi 2 Model B (armv7l)`
- Power supply used	| 5V 1A RAVpower
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Software title	| `wordpress`
- Was the software title installed freshly or updated/migrated? | `freshly`
- Can this issue be replicated on a fresh installation of DietPi? | `yes`
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. install dietpi
2. and install wp from soft-center

#### Expected behaviour:

- ...

#### Actual behaviour:

- "Your PHP installation appears to be missing the MySQL extension which is required by WordPress."

#### Extra details:

- ...

msongz, 2018-11-14:
`apt-get install php-mysql`
will fix

closed 2018-11-14 by msongz

Fourdee, 2018-11-14:
@msongz 

Many thanks for the report 👍 

---

Notes:
I was unable to replicate issue with current dev branch. ```php-mysql``` is installed as a pre-req.

MichaIng, 2018-11-14:
I guess it was due to: https://github.com/Fourdee/DietPi/issues/2217#issuecomment-435599804
- But that occurred only, if PHP was already installed previously with another database (e.g. SQLite) and then MySQL on a separate install run.
- So should not have appeared on a fresh install with selecting Wordpress (+others) all together.

msongz, 2018-11-15:
Yeah, I installed gitea binary with sqlite set up, not sure it is the reason

MichaIng, 2018-11-15:
@msongz 
Okay jep that makes sense then. Thanks for feedback, then indeed the issue is solved already.