MichaIng/DietPi #2245: Upmpdcli - MPD Upnp addon

potter-91, 2018-11-14:
UPMPDCLI
A UPnP front-end to MPD Server
https://www.lesbonscomptes.com/upmpdcli/
https://github.com/triplem/upmpdcli

Documentation at official website

Similar to GMediaRender but it is a addon to MPD
So if you are playing a MPD Playlist, it pauses the list when receiving a stream
GMR would play the stream additionally to what MPD is actually playing

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
Install MPD before ^^

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes
Yes, not so easy but I think I can handle it.

MichaIng, 2018-11-15:
@potter-91 
Thanks for your requests. There are some more on the pipe that I personally want to implement first, but we'll have a look.

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+19