MichaIng/DietPi #2246: How to update a dietpi software?

shlatchz, 2018-11-14:
Hello,

I received a notification in `Radarr` that a new version is available

[image]

How can I update `Radarr` (and `Sonarr`, `Jackett`, `Deluge` and any other software I installed with `dietpi-software`) software without breaking the whole dietpi setup?

Thanks!

shlatchz, 2018-11-15:
> * In case of Radarr, since is simply installed by downloading and unpacking an archive (https://api.github.com/repos/Radarr/Radarr/releases), you cannot break much. Although easiest is to `dietpi-software reinstall 145`

@MichaIng running `dietpi-software reinstall 145` won't delete the DB?

Fourdee, 2018-11-15:
@shlatchz 

Hi,

Radarr/Sonarr/Lidarr should be fine to reinstall via ```dietpi-software``` for updates from our side, as we do not touch any existing db files, previous one are left in place. This is of course assuming the latest Radarr/Sonarr/Lidarr binaries support the previous db files (which is the case in all my testing).

Jackett reinstall is currently non-functional in v6.17, fixed in v6.18: dietpi-software reinstall 145

Regardless, if in doubt, i'd highly recommend backing up the system, before-hand using:
```
dietpi-backup
```

msdos, 2018-11-20:
Should this information be somehow referenced in README.md?

Fourdee, 2018-11-20:
@msdos 

> Should this information be somehow referenced in README.md?

Yep, or Wiki/Forum link. We need to cover this somewhere 👍

ilgrank, 2019-02-03:
I'm running 6.20, and I've just installed Emby.
The installed version is 3.5.2.0, while latest is 4.0.1.0
is it normal DietPi installs an outdated version?
is it safe to upgrade? (via APT)?
Thanks!

bohtho, 2019-02-17:
Would this method also work on Node-RED ? I seem to remember it overwriting my configuration files and replacing the entire user directory the last time I tried to update Node-RED (via the GUI).

MichaIng, 2019-02-17:
@ilgrank 
DietPi-Software installs software from different sources. If an APT package is available, in most cases we simply install this and it is safe to upgrade via apt-get upadate/upgrade commands, or our non-interactive but error handled wrappers `G_AGUP` && `G_AGUG`.
But in some cases there are not APT packages available, in other cases they are very outdated, since Debian is very conservative in adding new releases. In practice major versions are frozen in Debian releases, only minor and security updates will be released through the Debian repo.

Emby server is not available on Debian repos at all, so we need to download manually from they GitHub source: https://github.com/MediaBrowser/Emby.Releases/releases
Meanwhile we found a way to automatically catch the latest release from: https://api.github.com/repos/MediaBrowser/Emby.Releases/releases/latest

With v6.22 `dietpi-software` will automatically install the latest upstream version based on the method above: https://github.com/Fourdee/DietPi/pull/2525

@bohtho 
Node-RED is installed via a totally different method. It is a (as can be seen by the name) Node.js module and installed via `npm`, the Node Package Manager.
You can update it accordingly via: `npm up -g node-red`
- `npm` the Node Package Manager binary
- `up` alias for the update command
- `-g` to handle the package globally (system wide) instead of installing/updating it within the current directory (locally).
- And the module/package names at the end.

You can also update all outdated globally installed Node modules (there are some Node-RED core modules that cannot be updated via the web UI), simply leave the module/package field empty: `npm up -g`

MichaIng, 2020-07-11:
I'm btw constantly adding "Update to newest version:" to your online docs, when touching those. We'll complement this when porting/copying those to our shiny new docs website: https://dietpi.com/docs/user-optimised-software/

closed 2020-07-11 by MichaIng