MichaIng/DietPi #2249: Email on Update

Micha-Btz, 2018-11-15:
At first, thanks for this great project.

Mostly the dietpi does this job without my attention, so it would be great, if it can send me a mail if a new update is available. Like apticron does.

Maybe I can create a script that test if there is an update available, is there a point where I can get the infos on the pi directly?

Thanks Micha

Fourdee, 2018-11-15:
@Micha-Btz 

Hi Micha,

Many thanks for the request 👍 

The main reason we do not offer this feature, is we simply do not log any contact or identifiable data on our users.
The only information we do obtain is hardware/software data:
https://dietpi.com/survey/

#### Current methods for update news:
- Twitter is probably the best bet, we post straight after release: https://twitter.com/dietpi_
- Forum: https://dietpi.com/phpbb/viewforum.php?f=10&sid=637a67be502be3ae925262c1a06a5435 (can be delayed)

---

If we receive more interest for this feature, we can look into allowing users to opt in for email notifications via the DietPi survey system.

MichaIng, 2018-11-15:
@Fourdee @Micha-Btz 
Of course it's possible to send an email vor local SMTP client, so credentials only need to be stored locally. This then also gives an update notification regardless of used Git branch, just parsing `/DietPi/dietpi/.update_available` flag 🙂.

Micha-Btz, 2018-11-15:
Oh, mail just locally from my pi to me. Not a global solution. I will make a test with parsing the /DietPi/dietpi/.update_available. I will report back.

Thanks

MichaIng, 2018-11-15:
`msmtp` seems to be the lightest solution. No full MTA (like exim e.g.) or GUI mailer:
- https://packages.debian.org/de/stretch/msmtp
- https://marlam.de/msmtp/msmtp.html
- Or install from source: https://marlam.de/msmtp/download/

Configuration looks quite clear, using TLS with `ca-certificates`. A random self-singed certificate should do. At least if, I would by default pre-configure TLS.

Only question is:
- How to send an email via CLI? We don't want/need MUTT or something like this, since we only send a simple plain text pre-written mail.

Micha-Btz, 2018-11-16:
I have a fully working postfix config on my pi, also send local mails with mailx and postfix send the mail out to my provider (smarthost).

So I only need a point that I can grep for a trigger and then cron sends me a mail. 
But maybe the msmtp solution is for other users as well.

mrmajik45, 2018-11-16:
So if it detects a new version of DietPi has been released it'll email them?

Micha-Btz, 2018-11-17:
this script should do what I want. If /DietPi/dietpi/.update_available is created, the script send me an email.
```
root@DietPi:~#
 cat /etc/cron.daily/update
#!/bin/bash
set -x
if test -f /DietPi/dietpi/.update_available;
     then mail -s "dietpi update available" your@email.com <<< "update available"
     echo "file not found""
     exit 0
fi
```

Micha-Btz, 2018-11-30:
For me it is solved, so I close this issue. If you want create a solution for all users, please reopen.

closed 2018-11-30 by Micha-Btz