MichaIng/DietPi #2250: Binary files can't be executed by command

FlameFireling, 2018-11-16:
I installed some software and when i tried to run it it gave unknown command error. Everytime i want to run a aplication i have to cd to the bin or games directory.

MichaIng, 2018-11-16:
@FlameFireling 
Welcome 🙂.

On GNU/Linux generally (actually the same with Windows), there is a `$PATH` variable that contains the folder, where the shell searches for binaries/commands to execute. By default these are:
- /bin
- /sbin
- /usr/bin
- /usr/sbin
- /usr/local/bin
- /usr/local/sbin

Software needs to place or link its binaries into one of those, to be executable within the shell without the need to enter the full path to the binary as well.
Some software also add their custom path to the $PATH variable, but this is actually not the cleanest method and against the idea of GNU/Linux standards.

So the question is how you installed the game. APT packages usually place/link their binaries automatically, so I guess you installed it from another source.
I would then link the needed binaries to the /usr/local/bin directory, which is meant to be used for manually installed software:
- `ln -s /path/to/game/binary /usr/local/bin/`

Now it should be executable from shell without the need to enter the full path.

FlameFireling, 2018-11-17:
I installed it using "apt-get install" from the raspbian repository.

MichaIng, 2018-11-18:
@FlameFireling 
Okay, usually APT packages should place their binaries correctly. Might I ask which software it is, so I can try to replicate? Perhaps there is something wrong with your $PATH variable.

Paste output of `echo "$PATH" && which -a which`

FlameFireling, 2018-11-19:
bsdgames, minetest and gimp

MichaIng, 2018-12-06:
@FlameFireling
Did you solve it meanwhile?

Fourdee, 2019-01-13:
Will mark as closed. If this is still an issue, please reopen.

closed 2019-01-13 by Fourdee