MichaIng/DietPi #2251: Minetest

FlameFireling, 2018-11-16:
Is an open source voxel game engine. Play one of our many games, mod a game to your liking, make your own game, or play on a multiplayer server. Would be great with PirateBox for portable server parties

MichaIng, 2018-11-16:
@FlameFireling 
Thanks for your request. Could you add some links to official websides, download links, install methods to your both requests? This will make it easier for us to check and implement.

MichaIng, 2018-12-26:
URL: https://www.minetest.net/
GitHub: https://github.com/minetest/minetest
Feel free to vote for it on FeatHub: https://feathub.com/Fourdee/DietPi/+20

MichaIng, 2019-02-27:
This request has been down voted to minus. Due to general cleanup, I will mark this issue as closed. However the FeatHub page stays open and as fast as this request reaches closer to top, we will reopen to start working on it.

closed 2019-02-27 by MichaIng