MichaIng/DietPi #2252: i3 window manager.

FlameFireling, 2018-11-16:
i3 is a tiling window manager, completely written from scratch. The target platforms are GNU/Linux and BSD operating systems. Support floating windows too

MichaIng, 2018-11-16:
https://github.com/Fourdee/DietPi/issues/2251

MichaIng, 2018-12-26:
URL: https://i3wm.org/
GitHub: https://github.com/i3/i3
Vote for it on FeatHub: https://feathub.com/Fourdee/DietPi/+21

FredericGuilbault, 2019-06-28:
AFAIK i3 is part of standard debian repository 

https://packages.debian.org/search?keywords=i3

MichaIng, 2019-06-28:
@FredericGuilbault 
Ah good find. Would be good to know, if this places itself as default window manager on install or one needs to configure it (which dietpi-software could do then). Most likely depends on desktop as well.

khru, 2020-06-27:
Will this feature be ready soon?

MichaIng, 2020-06-27:
Soon likely not, as there is much more on ToDo (as well many software titles with higher votes) but not enough time to implement them quickly. If anyone wants to help, this would speed the process of course: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

Since i3 is available via Debian repo, a start would be: `apt install i3`
See how it integrates into our supported desktops, LXDE, Xfce etc. or if it should be used as a window-manager-only X session.

The issue with desktop integrations is that Debian desktop packages mostly pull other window managers automatically:
- LXDE => Openbox: https://packages.debian.org/buster/lxde-core
- Xfce => xfwm4: https://packages.debian.org/buster/xfce4
- MATE => Macro: https://packages.debian.org/buster/mate-desktop-environment-core
- GNUStep at least recommends wmaker which we also install with it: https://packages.debian.org/buster/gnustep
- Only LXQt explicitly allows to install a window manager of own choice: https://packages.debian.org/buster/x-window-manager

So while in theory it should be possible to usw i3 on most desktops, I already read about incompatibilities with default desktop panels and you'll in most cases have two windows managers installed then. So it is not too trivial to implement a reasonable well working integration with all provided desktops. If i3 can be a desktop **replacement** (in case in combination with some other standalone applications) that would be an easier approach.

Silent-Hunter, 2021-02-28:
i3 is sort of like a minimal desktop environment. It includes its own panel, menu, notification manager, etc. so it should be pretty usable on its own. It is missing things like image viewer (I use sxiv), wallpaper (I use feh), document reader (I just use Atril from MATE), etc.

GordonZed, 2024-07-21:
i3 doesn't really cause conflicts the same way other DEs might. It's suuuper minimal. You can just install it from apt, and if you're in the terminal (tty, no X session), start it with:
`xinit i3 -device :1 -- :1 vt1`

(might need to change both occurrences of `:1` and `vt1` depending on your default tty... maybe? That or it may just require using `sudo` if you're targeting a new tty session