MichaIng/DietPi #2256: ODROID 3.5" LCD shield backlight not turning on with the current kernel

Kreeblah, 2018-11-18:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| v6.17.12
- Distro version	| stretch
- Kernel version	| Linux DietPi 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- SBC device		| Odroid C2 (aarch64)
- Power supply used	| https://ameridroid.com/products/5v-2a-power-supply
- SDcard used		| https://ameridroid.com/products/emmc-module-c2-linux-red-box

#### Additional Information (if applicable):
- Software title	| dietpi-cloudshell
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bugreport ID: dd8eaf40-8f9b-4377-8bbf-210dde45ce63

#### Steps to reproduce:

1. Perform a fresh install of DietPi
2. Configure DietPi to use the odroid-lcd35 screen via the display options in dietpi-config
3. Install dietpi-cloudshell
4. Configure dietpi-cloudshell to use tty1
5. Run dietpi-autostart to run dietpi-cloudshell at boot
6. Reboot

#### Expected behaviour:

- The LCD should light up and display the scenes from dietpi-cloudshell

#### Actual behaviour:

- The LCD doesn't light up.

#### Extra details:

- At some point, the paths to echo things to changed.  This is reflected in the ODROID documentation under how to enable the display on 18.04 at https://wiki.odroid.com/accessory/display/3.5inch_lcd_shield/3.5inch_lcd_shield
- I was able to resolve this by editing /etc/systemd/system/odroid-lcd35.service and removing the "/platform" strings in the paths where things get echoed to since that's the difference between the old and new paths.

MichaIng, 2018-11-20:
@Kreeblah @Fourdee 
Added PR for auto detection: https://github.com/Fourdee/DietPi/pull/2265
Theoretically it can still fail, when APT update to the new kernel version is done. Then user would need to re-enable the LCD without an LCD (e.g. after reboot). Not sure if we can assume that on Jessie the config files remain the same and on Stretch, with next dietpi-update, APT updates the kernel accordingly?

Then we could use config files based on G_DISTRO and simply update the service file for all C1/2 Stretch systems on patch_file.

Fourdee, 2018-11-20:
```
root@DietPi:~# find /sys -name pwm*
/sys/bus/platform/devices/pwm
/sys/bus/platform/devices/pwm-ctrl
/sys/devices/pwm
/sys/devices/pwm-ctrl
```

MichaIng, 2018-11-20:
@Fourdee 
Do you have a chance to check on Jessie?

Fourdee, 2018-11-20:
@MichaIng 

1 sec, fails on install due to not modprobe, dir not available.

Fourdee, 2018-11-20:
Testing:
- 🈯️ C2 Stretch
- Jessie

MichaIng, 2018-11-20:
@Fourdee 
Ah makes sense. Is anyway food to use modprobe where possible, so we can start service directly skip reboot requirement as well?

Fourdee, 2018-11-20:
@MichaIng 

> so we can start service directly skip reboot requirement as well?

In theory yep 👍

Fourdee, 2018-11-20:
Jessie working fine 👍 

```
root@DietPi:~# cat /etc/systemd/system/odroid-lcd35.service
[Unit]
Description=Odroid LCD 3.5

[Path]
PathExists=/dev/fb2

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/bin/bash -c 'echo 500000 > /sys/devices/platform/pwm-ctrl/freq0'
ExecStart=/bin/bash -c 'echo 1 > /sys/devices/platform/pwm-ctrl/enable0'
ExecStart=/bin/bash -c 'echo 1023 > /sys/devices/platform/pwm-ctrl/duty0'
ExecStart=/usr/bin/setterm -blank 0 -powersave off
ExecStart=/bin/con2fbmap 1 2
StandardOutput=tty

[Install]
WantedBy=multi-user.target
root@DietPi:~# ls -lha /sys/devices/platform/pwm-ctrl/freq0
-rwxrwxrwx 1 root root 4.0K Nov 20 20:11 /sys/devices/platform/pwm-ctrl/freq0
```

Completed. Thanks @MichaIng

closed 2018-11-20 by Fourdee