MichaIng/DietPi #2257: PREP_SYSTEM_FOR_DIETPI.sh fails because https is enabled in sources.list

optio50, 2018-11-18:
PREP_SYSTEM_FOR_DIETPI.sh
Method https has died unexpectedly
fails because sources.list https is enabled before during and after install of DietPi
Hardware is Odroid U3 with https://oph.mdrjr.net/meveric/images/Jessie/Debian-Jessie-1.1-20170515-U.img.xz
Debian Jessie & DietPi upgrades to Stretch
Kernel 3.8.13.30

I was able to complete the install by simultaneously starting another login and modifying the /etc/apt/sources.list while the script was running.

MichaIng, 2018-11-18:
@optio50 
Thanks for your report.

Ahh I remember seeing this error some time ago, but cannot find related issue or remember how we solved it.

- Can you paste the related part of the terminal output, to let us see on which APT call this happened?
- Can you verify `apt-transport-https` and `ca-certificates` packages are installed: `dpkg-query -s apt-transport-https ca-certificates`

> fails because sources.list https is enabled before during and after install of DietPi

So this error appears before even running DietPi-PREP script? Test on fresh Meveric image and then: `apt update && apt full-upgrade`

> I was able to complete the install by simultaneously starting another login and modifying the /etc/apt/sources.list while the script was running.

Jep, best solution/attempt whenever a DietPi scripts throws an error and prompts a whiptail menu, since you can mostly choose "retry".

But in this case would be good to find the reason, since we want APT to use `https` whenever available for security and privacy reasons. With `apt-transport-https` and `ca-certificates` packages this should generally work.

Fourdee, 2018-11-18:
Strange, Jessie is set to http in DietPi code:
- https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_software#L158-L167

Only enabled for Stretch+
- https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L865-L870

Fourdee, 2018-11-18:
> & DietPi upgrades to Stretch

Ah yes, I see. 

```DISTRO_TARGET=stretch```, and ```G_DISTRO=$DISTRO_TARGET G_DISTRO_NAME="$DISTRO_TARGET_NAME" G_HW_MODEL=$G_HW_MODEL G_RUN_CMD /DietPi/dietpi/func/dietpi-set_software apt-mirror 'default'``` before pre-req packages are installed ```apt-transport-https```.

Fourdee, 2018-11-18:
Pre-req ```apt-transport-https``` at start of script should fix it, requires testing to confirm:
https://github.com/Fourdee/DietPi/commit/d5dca2512d98412a66892585e582cd9e51b0776b

Dev branch PREP with updated code:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```

@optio50 
Any chance you would be willing to re-try PREP on  https://oph.mdrjr.net/meveric/images/Jessie/Debian-Jessie-1.1-20170515-U.img.xz using the above updated code and see if issue is resolved?

optio50, 2018-11-18:
Fresh install

Fails at Step 5

DietPi-PREP: G_AGI: apt-transport-https apt-utils bash-completion bc bzip2                                                                                        
│ ca-certificates console-setup cron curl debconf dirmngr ethtool fake-hwclock gnupg                                                                                
│ htop iputils-ping isc-dhcp-client kmod locales nano p7zip-full parted psmisc                                                                                      
│ resolvconf sudo systemd-sysv tzdata udev unzip usbutils wget whiptail crda
| firmware-atheros firmware-brcm80211 firmware-iwlwifi iw rfkill wireless-tools                                                                                     
│ wpasupplicant firmware-ralink dropbear initramfs-tools dosfstools hdparm                                                                                          
│  - Exit code: 100                                                                                                                                                 
│  - DietPi version: v6.16.0 (Fourdee/master) | HW_MODEL:13 | HW_ARCH:2 | DISTRO:4                                                                                  
│  - Image creator: n/a                                                                                                                                             
│  - Pre-image: n/a                                                                                                                                                 
│                                                                                                                                                                   
│ Log file contents:                                                                                                                                                
│ E: Method https has died unexpectedly!                                                                                                                            
│ E: Sub-process https received a segmentation fault.                                                                                                               
│ E: Method /usr/lib/apt/methods/https did not start correctly 

I logged in a separate console and disabled https in /etc/apt/sources.list and hit retry.

It continues for a few more min and then again

DietPi-Set_Hardware: G_AGI: alsa-utils                                                                                                                            
│  - Exit code: 100                                                                                                                                                 
│  - DietPi version: v6.17.0 (Fourdee/master) | HW_MODEL:13 | HW_ARCH:2 | DISTRO:4                                                                                  
│  - Image creator: n/a                                                                                                                                             
│  - Pre-image: n/a                                                                                                                                                 
│                                                                                                                                                                   
│ Log file contents:                                                                                                                                                
│ E: Method https has died unexpectedly!                                                                                                                            
│ E: Sub-process https received a segmentation fault.                                                                                                               
│ E: Method /usr/lib/apt/methods/https did not start correctly                                                                                                      
│                                                                                                                                                                   
│                                                                                                                                                                   
│                                                                                                                                                                   
│ Unable to continue, DietPi-Set_Hardware will now terminate.    

https was reenabled so I then again disable https in /etc/apt/sources.list and hit retry. 

Script completes at this point.

reboot
first login

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
-bash: [: missing `]'
-bash: [: missing `]'
-bash: [: missing `]'
-bash: [: missing `]'

apt update

Get:1 http://fuzon.co.uk/meveric all InRelease [15.2 kB]
Get:2 http://fuzon.co.uk/meveric jessie InRelease [5630 B]
Get:3 http://fuzon.co.uk/meveric all/main armhf Packages [39.4 kB]
Get:4 http://fuzon.co.uk/meveric all/testing armhf Packages [7420 B]
[... cut ...]
0% [Working]

https was reenabled so I then again disable https in /etc/apt/sources.list 

apt update completes correctly

reboot

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
-bash: [: missing `]'
-bash: [: missing `]'
-bash: [: missing `]'
-bash: [: missing `]'

optio50, 2018-11-19:
Well after several more attempts I still cant get the revised PREP_SYSTEM_FOR_DIETPI.sh
So I locked down the  sources.list with chattr +i and it completed successfully. However, still has erros and the banner never comes up.

So I left the chattr +i in place  and retried with the original published git PREP_SYSTEM_FOR_DIETPI.sh
Complet successfully rebooted proceded with minimal install and all seems to be ok. Currently install additional software. Well see how it goes (also removing the chattr +i now)

MichaIng, 2018-11-19:
@optio50 
This means Pi-hole did not install successfully, right? Otherwise the file would be in place. Looks like the installer failed.
I guess it should, if APT fails, as it also does+requires some APT package installs. But to verify that, we would require install log. Pi-hole should show about what failed an what succeeded.

MichaIng, 2018-11-19:
@optio50 
Seems to be.

The part before this messages is the interesting one, the ones you posted are expected, depending on how much time the APT install (attempt) took.

As said I faced this error once and it required a special solution. Have to do some research to find the needed hints/remember what it was.

MichaIng, 2018-11-19:
@optio50 
I understand. 

Did more research and indeed the error often appears due to faulty packages. So either apt-transport-https or the apt package itself seem to be faulty.

Solution could be to switch to http:// sources and reinstall both packages: 
apt install --reinstall apt apt-transport-https

Others also reported that it could only be fixed by purging the pachage(s) first and do a fresh install.
But the error might be in the base image already. To test, before running DietPi-PREP,  install the https package, switch and test https deb sources.

But this is something we can test, if @Fourdee has an U3 there?

Fourdee, 2018-11-19:
@MichaIng 

>  @Fourdee has an U3 there?

Nope ☹️. Although, could try replicating on XU4 jessie image, will have a stab when I get some time.

MichaIng, 2019-02-27:
Will mark as closed due to general cleanup. Feel free to reopen if issue persists with current version.

closed 2019-02-27 by MichaIng