MichaIng/DietPi #2261: Pinebook 1080p

Fourdee, 2018-11-19:
Notes:

#### Ayufan
```
root@DietPi:~# lsmod
Module                  Size  Used by
vfe_v4l2              772828  0
uvcvideo               79606  0
videobuf2_vmalloc      13221  1 uvcvideo
videobuf2_dma_contig    20341  1 vfe_v4l2
videobuf2_memops       12600  2 videobuf2_vmalloc,videobuf2_dma_contig
vfe_io                 44013  1 vfe_v4l2
videobuf2_core         35245  2 uvcvideo,vfe_v4l2
ss                     41661  0
cedar_ve               20392  0
mali                  287546  0
8723cs               1227342  0
8723bs               1496864  0
sunxidrm             1006550  1
drm_kms_helper         39436  1 sunxidrm
hall                   14030  0
hci_uart               31036  0
bluetooth             217400  1 hci_uart
8723bs_vq0           1501780  0
cfg80211              430891  3 8723bs_vq0,8723bs,8723cs
ip_tables              22654  0
x_tables               26209  1 ip_tables
```

```
root@DietPi:~# cat /etc/udev/rules.d/90-sunxi-disp-permission.rules
KERNEL=="disp", MODE="0770", GROUP="video"
KERNEL=="cedar_dev", MODE="0770", GROUP="video"
KERNEL=="ion", MODE="0770", GROUP="video"
KERNEL=="mali", MODE="0770", GROUP="video"
```

---

#### Neon

enable display
```
modprobe sunxidrm
```

```
root@DietPi:~# dmesg | grep drm
[    8.833038] [drm] Initialized drm 1.1.0 20060810
[   12.243973] [drm:sunxi_drm_load] *ERROR* [dengbo] debug from here
[   12.251676] [drm:sunxi_drm_get_name_node] *ERROR* of_find_compatible_node allwinner,sunxi-lcd1 fail
[   12.261592] [drm:disp_sys_script_get_item] *ERROR* get [lcd1] item err.
[   12.269285] [drm:sunxi_drm_get_name_node] *ERROR* of_find_compatible_node allwinner,sunxi-lcd1 fail
[   12.284168] [drm:disp_sys_script_get_item] *ERROR* get [lcd1] item err.
[   12.291857] [drm:sunxi_drm_get_name_node] *ERROR* of_find_compatible_node allwinner,sunxi-lcd1 fail
[   12.302992] [drm:disp_sys_script_get_item] *ERROR* get [lcd1] item err.
[   12.312505] [drm] [1048] crtc_id:0(22)
[   12.316635] [drm] [1048] crtc_id:1(35)
[   12.330987] [drm] [688] encoder_id:0(36) possible_crtcs:0x3
[   12.337091] [drm] [688] encoder_id:1(37) possible_crtcs:0x3
[   12.343719] [drm:sunxi_hdmi_pan_init] *ERROR* status for [sunxi-hdmi] is not 'okay': disabled
[   12.353457] [drm] failed to get [lcd_fix_power] string.
[   12.359685] [drm] failed to get [lcd_fix_power1] string.
[   12.368211] [drm] failed to get [lcd_fix_power2] string.
[   12.376744] [drm] failed to get [lcd_fix_power3] string.
[   12.383403] [drm] failed to get [lcd_power3] string.
[   12.388852] [drm] failed to get gpio[lcd_gpio_2].
[   12.393998] [drm] failed to get gpio[lcd_gpio_3].
[   12.399135] [drm] failed to get gpio[lcd_gpio_4].
[   12.410184] [drm] failed to get gpio[lcd_gpio_5].
[   12.415329] [drm] failed to get gpio[lcd_gpio_scl].
[   12.420659] [drm] failed to get gpio[lcd_gpio_sda].
[   12.420668] [drm] failed to get [lcd_gpio_power0] string.
[   12.420672] [drm] failed to get [lcd_gpio_power1] string.
[   12.420676] [drm] failed to get [lcd_gpio_power2] string.
[   12.420683] [drm] failed to get [lcd_pin_power1] string.
[   12.420687] [drm] failed to get [lcd_pin_power2] string.
[   12.421038] [drm] failed to get [lcd_size] string.
[   12.421461] [drm] [556]sunxi_con_id:0(38), fix:0
[   12.421675] [drm] Supports vblank timestamp caching Rev 1 (10.10.2010).
[   12.421678] [drm] No driver support for vblank timestamp query.
[   12.442534] [drm] connetor(0)--->crtc(0) encoder:36,resolution(1920 x 1080)
[   12.442543] [drm] surface(1920 x 1080), fb(1920 x 1080) and bpp(32) form(875713089)
[   12.509494] [drm] set parent clk:777000000, but get:779625000.
[   12.509513] [drm] set clk:777000000, but get:779625000.
[   12.511134] [drm] sunxi_drm_sys_gpio_request, gpio_name=lcd_gpio_0, gpio=120, <1,-1,0,0>ret=0
[   12.511175] [drm] sunxi_drm_sys_gpio_request, gpio_name=lcd_gpio_1, gpio=230, <1,-1,0,1>ret=0
[   12.511289] [drm] suceess to enable regulator vcc-pd!
[   12.512351] sunxidrm edp_panel_open
[   12.512497] [drm] suceess to enable regulator vcc-lvds!
[   12.512615] [drm] suceess to enable regulator vcc-mipi!
[   12.512710] [drm] suceess to enable regulator vcc-hsic-12!
[   12.838215] [drm] sunxi_drm_sys_gpio_request, gpio_name=lcd_bl_en, gpio=119, <1,-1,0,1>ret=0
[   12.992056] sunxi-drm sunxi-drm: fb0:  frame buffer device
[   12.992060] sunxi-drm sunxi-drm: registered panic notifier
[   12.992071] [drm] Initialized sunxi 1.0.0 20160530 on minor 0
```

```xinit startx```
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep EE
[    90.103] Current Operating System: Linux DietPi 3.10.105-bsp-1.2 #2 SMP PRE MPT Sat Oct 27 19:24:05 IST 2018 aarch64
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    90.245] (EE) ARMSOC(0): ERROR: Pixman failed to blit from /dev/fb0 to scanout buffer
[    90.250] (EE) MODE :0
[    90.279] (EE) AIGLX error: dlopen of /usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so failed (/usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so: cannot open shared object file: No such file or directory)
[    90.279] (EE) AIGLX: reverting to software rendering
[    91.656] (EE) MODE :0
[    92.904] (EE) MODE :0
[    92.904] (EE) MODE :0
[    93.905] (EE) MODE :0
[    93.905] (EE) MODE :0
```

```startx```
```
[   200.704] Current Operating System: Linux DietPi 3.10.105-bsp-1.2 #2 SMP PRE MPT Sat Oct 27 19:24:05 IST 2018 aarch64
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   200.796] (EE) ERROR: Cannot set the DRM interface version.
[   200.796] (EE) ERROR: Cannot open a connection with the DRM - Permission denied
[   200.796] (EE) No devices detected.
[   200.796] (EE)
[   200.796] (EE) no screens found(EE)
[   200.796] (EE)
[   200.796] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   200.796] (EE)
[   200.823] (EE) Server terminated with error (1). Closing log file.
```

```
root@DietPi:~# lsmod
Module                  Size  Used by
vfe_v4l2              793389  0
uvcvideo               87884  0
ax88179_178a           23196  0
videobuf2_vmalloc      13221  1 uvcvideo
usbnet                 32975  1 ax88179_178a
videobuf2_dma_contig    20341  1 vfe_v4l2
videobuf2_memops       12600  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_core         43437  2 uvcvideo,vfe_v4l2
vfe_io                 48109  1 vfe_v4l2
cedar_ve               24488  0
ss                     45575  0
mali                  328420  0
8723cs               1373116  0
8723bs               1666782  0
cfg80211              475949  2 8723bs,8723cs
hall                   14030  0
sunxidrm             1035674  1
drm_kms_helper         39490  1 sunxidrm
ip_tables              22607  0
x_tables               30305  1 ip_tables
```

Fourdee, 2018-11-19:
#### Ayufan 2 / Buster:

https://github.com/ayufan-pine64/linux-build/releases/download/0.7.30/linux-pine64-package-0.7.30.deb
```
linux-pine64-package
```

Battery:
```
cat /sys/class/axppower/power_sply
```

---

#### GPU
```
root@DietPi:~# find / -name armsoc*
root@DietPi:~#
```

Fourdee, 2018-11-20:
```60-xserver-xorg-video-armsoc-sunxi.rules```
```
# Set the right permissions to the devices provided by the
# armsoc driver
ENV{ACL_MANAGE}=="0", GOTO="mali_gpu_end"
ACTION!="add|change", GOTO="mali_gpu_end"

# graphics devices
ACTION=="add|change", KERNEL=="mali*", TAG+="uaccess", TAG+="udev-acl"

LABEL="mali_gpu_end"
```

ayufan, 2018-11-20:
You cannot add `sunxidrm` when `disp`/`hdmi` is loaded. You should blacklist the `disp`/`hdmi` before.

Fourdee, 2018-11-20:
@ayufan 

Thank you 👍 Trying it now.

Notes:
```
mv /etc/modules-load.d/pine64-sunxidrm.conf.disabled /etc/modules-load.d/pine64-sunxidrm.conf
cat << _EOF_ > /etc/modprobe.d/blacklist-hdmi.conf
hdmi
disp
_EOF_
```

```
[    40.890] (EE) systemd-logind: failed to take device /dev/dri/card0: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
root@DietPi:~# systemctl unmask systemd-logind
```

Fourdee, 2018-11-22:
Disabled xorg.conf's:
- 🈴 ```sunxidrm``` 
- 🈴 ```hdmi + disp```

```
Section "Device"
        Identifier      "Allwinner A10/A13 FBDEV"
        Driver          "fbdev"
        Option          "fbdev" "/dev/fb0"
        Option          "Backlight" "lcd0"

       # Option          "SwapbuffersWait" "true"
EndSection
```
- 🈴 ```sunxidrm```  | black screen
```
[    32.767] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
```
- 🈴 ```hdmi + disp``` 
```
root@DietPi:~# cat /var/log/Xorg.0.log
[    40.596]
X.Org X Server 1.19.2
Release Date: 2017-03-02
[    40.596] X Protocol Version 11, Revision 0
[    40.596] Build Operating System: Linux 4.9.0-8-arm64 aarch64 Debian
[    40.596] Current Operating System: Linux DietPi 3.10.105-bsp-1.2-ayufan-140 #1 SMP PREEMPT Tue Oct 30 14:21:35 UTC 2018 aarch64
[    40.596] Kernel command line: console=ttyS0,115200n8 enforcing=0 cma=384M no_console_suspend androidboot.serialno=941078878918602207cf androidboot.hardware=sun50iw1p1 androidboot.selinux=permissive earlyprintk=sunxi-uart,0x01c28000 loglevel=8 root=/dev/mmcblk0p2 eth0_speed=auto
[    40.597] Build Date: 03 November 2018  03:03:37AM
[    40.597] xorg-server 2:1.19.2-1+deb9u5 (https://www.debian.org/support)
[    40.597] Current version of pixman: 0.34.0
[    40.598]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    40.598] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    40.598] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Nov 22 17:41:03 2018
[    40.599] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    40.599] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    40.600] (==) ServerLayout "ServerLayout0"
[    40.600] (==) No screen section available. Using defaults.
[    40.600] (**) |-->Screen "Default Screen Section" (0)
[    40.600] (**) |   |-->Monitor "<default monitor>"
[    40.600] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    40.600] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    40.600] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    40.600] (**) Option "BlankTime" "0"
[    40.600] (**) Option "StandbyTime" "0"
[    40.600] (**) Option "SuspendTime" "0"
[    40.600] (**) Option "OffTime" "0"
[    40.600] (==) Automatically adding devices
[    40.600] (==) Automatically enabling devices
[    40.600] (==) Automatically adding GPU devices
[    40.601] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    40.601] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    40.601]    Entry deleted from font path.
[    40.601] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    40.601]    Entry deleted from font path.
[    40.601] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    40.601]    Entry deleted from font path.
[    40.601] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    40.601]    Entry deleted from font path.
[    40.601] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    40.601]    Entry deleted from font path.
[    40.601] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    40.601]    Entry deleted from font path.
[    40.601] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
[... 210 lines cut ...]
[    42.515] (**) Option "fd" "21"
[    42.515] (**) Option "fd" "22"
[    42.516] (**) Option "fd" "23"
[    42.516] (**) Option "fd" "24"
[    42.516] (**) Option "fd" "25"
[    42.516] (**) Option "fd" "26"
[    42.517] (**) Option "fd" "22"
[    42.519] (II) UnloadModule: "libinput"
[    42.519] (II) systemd-logind: not releasing fd for 13:69, still in use
[    42.519] (II) UnloadModule: "libinput"
[    42.519] (II) systemd-logind: releasing fd for 13:70
[    42.660] (II) UnloadModule: "libinput"
[    42.660] (II) systemd-logind: releasing fd for 13:64
[    42.780] (II) UnloadModule: "libinput"
[    42.780] (II) systemd-logind: releasing fd for 13:66
[    42.890] (II) UnloadModule: "libinput"
[    42.890] (II) systemd-logind: releasing fd for 13:71
[    43.010] (II) UnloadModule: "libinput"
[    43.010] (II) systemd-logind: releasing fd for 13:69
[    43.130] (II) UnloadModule: "libinput"
[    43.130] (II) systemd-logind: releasing fd for 13:68
[    43.251] (II) UnloadModule: "libinput"
[    43.251] (II) systemd-logind: releasing fd for 13:65
[    43.410] (II) Server terminated successfully (0). Closing log file.
```

Fourdee, 2018-11-22:
```lightdm``` (previously ```startx```)
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep EE
[   353.953] Current Operating System: Linux DietPi 3.10.105-bsp-1.2-ayufan-140 #1 SMP PREEMPT Tue Oct 30 14:21:35 UTC 2018 aarch64
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   353.991] (EE) ARMSOC(0): ERROR: Cannot set the DRM to return all planes.
[   354.023] (EE) ARMSOC(0): ERROR: Pixman failed to blit from /dev/fb0 to scanout buffer
[   354.053] (EE) AIGLX error: dlopen of /usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so failed (/usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so: cannot open shared object file: No such file or directory)
[   354.053] (EE) AIGLX: reverting to software rendering
```
- 🈴 login = black screen

Fourdee, 2018-11-22:
#### Try kernel+uboot from KDE neon image with DietPi RootFS:
```
8723bs
sunxidrm
```
- Hide wlan1

Fourdee, 2018-11-22:
So, results:
@ayufan 
- Sound works on your kernel 👍 
- ```hdmi+disp``` fails with ```[   38.907508] [DISP] disp_ioctl,line:1622:para err in disp_ioctl, cmd = 0x0,screen id = 65536```, under ```fbdev``` and ```fbturbo```.
- Unable to get ```armsoc``` working currently with Debian, however, ```fbturbo``` does work using the kde neon kernel.

Fourdee, 2018-11-28:
EMMC install via SD:
```
wget https://dietpi.com/downloads/images/DietPi_Pinebook-ARMv8-Stretch-Desktop.7z
7z x *.7z
dd if=$(find DietPi*.img) of=/dev/mmcblk2
```

Fourdee, 2018-11-28:
Image now live: 
https://twitter.com/DietPi_/status/1067860201158651904

Completed.

closed 2018-11-28 by Fourdee