MichaIng/DietPi #2266: META | Cleanup obsolete branches/files

MichaIng, 2018-11-20:
@Fourdee 
As a cleanup step, I merged the `dietpi-fan_control` script from it's own branch into main line `.meta` folder, also to have it a bid more prominent. I am still aiming to work on this by times 😉.

What do you think about removing the obsolete Wheezy branch as well? Old version can theoretically always accessed/recovered from commit history, so this only has affect on running Wheezy system update checks. Dropping the branch completely underlines this being dropped from support.

Also I searched through GitHub branch and dietpi.com forum for references/links to testing branch and TESTING-BRANCH.md and changed all I found to dev branch respectively BRANCH_SYSTEM.md.
So we could theoretically removed both?
Removing testing branch would then only affect systems update checks, that are still running on testing branch. But these fail as well, since we did not merge the new branch+pre-patch system into that branch. Updates are doomed to fail currently and I believe there are no users on that branch anyway, if we see the advertised beta branch numbers 😉.

MichaIng, 2018-11-20:
Another thing: GitHub page header still shows: `Lightweight justice for your single-board computer. http://dietpi.com`
- Note http instead of https 😉. Minor one but should be easy to fix as well.

Fourdee, 2018-11-20:
@MichaIng 

> Note http instead of https 😉. Minor one but should be easy to fix as well.

Good spot, although we have a redirect via certbot! 😃

MichaIng, 2018-11-20:
@Fourdee 
> Good spot, although we have a redirect via certbot! 😃

Jep, visual only 😄.

MichaIng, 2019-02-27:
Done.

closed 2019-02-27 by MichaIng