MichaIng/DietPi #2267: DietPi-Software | Grafana: Passwords with special chars are added wrong

MichaIng, 2018-11-20:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=15&t=5248&p=15446#p15446

MichaIng, 2018-11-20:
Solved via: https://github.com/Fourdee/DietPi/commit/1fb6414aa26e3ac3f2017e72b942bc6b753181b6
- Whoopsie, actually wanted to add this via PR. However fix is very straight forward according to docs, works with simple PWs as well.

Fourdee, 2018-11-20:
Install + use test
- ARMv8 | pw=dietpi
```
[  OK  ] DietPi-Software | Comment in /etc/grafana/grafana.ini converted to setting: admin_password = """dietpi"""
```
-- GUI login 🈯️

Fourdee, 2018-11-20:
Install + use test:
- ARMv8 | pw =$&\"moocow
-- 🈯️ GUI login 
```
[  OK  ] DietPi-Software | Comment in /etc/grafana/grafana.ini converted to setting: admin_password = """$&\"moocow"""
```

Fourdee, 2018-11-20:
@MichaIng 

Testing passed, great work 👍 🥇

Fourdee, 2018-11-20:
Patch?

Probably best to manually uninstall + reinstall for affected users?

MichaIng, 2018-11-20:
@Fourdee 
Patch not required from my point of view.

Users who have this installed are either not affected or fixed it manually, obviously :wink:.

So this is only important for fresh installs when global password is applied.

Fourdee, 2018-11-25:
@MichaIng 

Great work, local testing passed, will mark as completed 👍

closed 2018-11-25 by Fourdee