MichaIng/DietPi #2269: Can some ideas from AtoMiC-ToolKit be used in DietPi?

msdos, 2018-11-21:
Don't know why but I came across https://github.com/htpcBeginner/AtoMiC-ToolKit

> AtoMiC Toolkit simplifies HTPC / Home Server setup and management on Ubuntu and Debian variants including Raspbian. It currently supports: Couchpotato, Deluged, Emby, FFmpeg, Headphones, Htpcmanager, Jackett, Kodi, Lazylibrarian, Madsonic, Mono, Mylar, Nzbget, Nzbhydra, NzbToMedia, Ombi, Plex, Plexpy, Pyload, qBittorrent, Radarr, Sabnzbd+, Sickg… https://www.smarthomebeginner.com

Maybe some ideas of this project can be used in DietPi, so I posting it here. However, I don't suggest installing both since dependency management can become a real nightmare.

Fourdee, 2018-11-21:
@msdos 

Interesting project, thanks for sharing 👍 

Generally speaking, with DietPi we like to do things "our own way" and find the best way of achieving it through testing and available knowledge.
I personally try to avoid looking at others projects, try to figure it out in our own way.

Regardless, the project looks interesting. Similar goals of the projects as DietPi, but very different approaches.

MichaIng, 2018-11-23:
Sadly looks dead, some definitely broken installers (Sickrage 😄) and such, no commit for half a year (besides meta).

What I like is the backup&restore for all apps, this is something I also wanted to add for more apps.
But stuck after Next/ownCloud 😆: https://github.com/Fourdee/DietPi/blob/master/.meta/dietpi-cloud-migration

What they use and I find helpful as well: https://feathub.com/
Makes it easier for our users to support/vote for image/software/feature requests and a better overview for us?

MichaIng, 2018-11-23:
Ah okay, jep much easier to maintain, but also less useful, since starting with docker and install apps there is already quite easy, AFAIK. Not too much benefit to have another helper tool layer on top from my point of view 😄.

Yeah all kind of isolation/virtualization/container solutions is the opposite of what DietPi is aiming for: Native bare integration of software titles into Linux for highest performance and least resource usage + dependencies. Also there it is most helpful to have an automated installer, since most, by times platform dependant, steps required.

msdos, 2018-11-28:
I'm closing this issue since you now know this alternative exists.

MichaIng, 2018-12-06:
@Fourdee
What you think about using: https://feathub.com/

closed 2018-12-25 by MichaIng