MichaIng/DietPi #2270: fbtft missing

troyane, 2018-11-21:
## Creating a bug report/issue:

I'm writing this since I can't find any related issue and/or forum thread.

Execution of next command shows that fbtft module is missing. Isn't it an essential part of Linux for 4.14 kernel?

```
# modinfo fbtft ;  modinfo fbtft_device
modinfo: ERROR: Module fbtft not found.
modinfo: ERROR: Module fbtft_device not found.
```

#### Required Information:
- DietPi version	| `G_DIETPI_VERSION_CORE=6 G_DIETPI_VERSION_SUB=17 G_DIETPI_VERSION_RC=12 G_GITBRANCH=master G_GITOWNER=Fourdee`
- Distro version	| `strech 9.6`
- Kernel version	| `Linux DietPi 4.14.52 #19 SMP Thu Oct 11 11:33:51 CST 2018 armv7l GNU/Linux`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used	| 5V 1A RAVpower
- SDcard used		| SanDisk ultra

Fourdee, 2018-11-21:
@troyane 

Hi,

Which device are you running?

Fourdee, 2018-11-24:
@troyane 

Appears the kernel module is missing from the official FriendlyARM pre-image we used.

Have you had any success with ARMbian by removing the ```network-manager``` package, setting up WiFi via ```/etc/network/interfaces```, then running DietPi PREP?
https://github.com/Fourdee/DietPi/issues/2210#issuecomment-435687637

---

Additional REF:
- https://github.com/Fourdee/DietPi/issues/2100

troyane, 2018-11-26:
@Fourdee your suspicions came true. On newly installed Armbian without network-manager and after successful execution of  DietPi PREP there was `fbtft`, see screenshot attached:
[image]

---
Correct me if I'm wrong -- you've used nanopi-neo-air_sd_debian-nas-jessie_4.14_armhf_20180919 (from accessory-ROMs folder on [official Google Drive](https://drive.google.com/drive/folders/1DHc5Pnrj8O8sU4McoPBgCgubskZ_UKiT)) since it was only one officially supported Debian-based (not Ubuntu-based) distro. 

Don't you think that it will be much better/easier to use Armbian distro as a base instead?

Fourdee, 2018-11-26:
@troyane 

> Don't you think that it will be much better/easier to use Armbian distro as a base instead?

Yes, we previously offered ARMbian based images for most of our SBC's in the past. However, things did not work out. 

Maybe in the future that might change, but for now we only use official FriendlyARM pre-images (+kernel).

MichaIng, 2020-03-03:
@troyane 
A new image is based on Armbian again and should solve the issue: https://dietpi.com/downloads/testing/DietPi_NanoPiNEOAir-ARMv7-Buster.7z

MichaIng, 2020-06-18:
Module is available on the new image, I mark this as closed.

closed 2020-06-18 by MichaIng