MichaIng/DietPi #2275: TinkerOS_Debian V2.0.8

ibnbd, 2018-11-23:
Tinker OS has new version, as I can see the beta at `v6.18.9`. Hopefully you will consider adding latest TinkerOS for `v6.18` final release.

Thank You

Fourdee, 2018-11-25:
@ibnbd 

Thanks, we'll check and redo image if its running a earlier tinkerOS version.

ibnbd, 2018-11-25:
@Fourdee thank you very much

Fourdee, 2018-11-30:
Currently 2.0.7, due to WiFi issues, will retry with 2.0.8.

Fourdee, 2018-11-30:
Image live:
https://dietpi.com/download

Completed.

closed 2018-11-30 by Fourdee

Millichrome, 2019-03-27:
Sorry for necroing this. 

The SSH login data still says 2.0.5. 

https://i.imgur.com/5HiE6rP.jpg

Btw, thank you for the new release! DietPi is the best for DIY home servers!

MichaIng, 2019-03-27:
@Millichrome 
Did you install the fresh image? Updating DietPi will not raise the underlying Tinker OS version AFAIK, though of course all APT packages etc. are updated.
But especially the `/DietPi/dietpi/.prep_info` file which provides the info for the banner is not updated. This just says which base image was used at the time the DietPi image was created.

Millichrome, 2019-03-27:
Ah, I see. It's an updated install on my "production" home server.

Thanks for the clarification.