MichaIng/DietPi #2283: Nexcloud Fails to Install

joaofl, 2018-11-25:
#### Details:
- Date           | Sun 25 Nov 15:02:50 WET 2018
- Bug report     | N/A
- DietPi version | v6.17.12 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07)
- Distro         | stretch (index=4)
- Command        | ncc maintenance:install
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:
Simply run the installation scrip from the dietpi-software menu.

#### Expected behaviour:
Should install and get it running 

#### Actual behaviour:
Failing in the middle.

#### Additional logs:
```
Log file contents:
An unhandled exception has been thrown:
Error: Class 'Memcached' not found in /var/www/nextcloud/config/config.php:1156
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(214): include()
#1 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#2 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#4 /var/www/nextcloud/lib/base.php(1068): OC::init()
#5 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#6 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#7 {main}
```

MichaIng, 2018-11-25:
@joaofl 
Thanks for report.

However please search for existing issues, there are about 20 of the exactly same error across GitHub and dietpi.com forum now 😉: https://github.com/Fourdee/DietPi/issues/2279

closed 2018-11-25 by MichaIng

joaofl, 2018-11-25:
I actually did search, but not found it :open_mouth:

joaofl, 2018-11-25:
I searched for the open issues only :sob:

MichaIng, 2018-11-25:
@joaofl 
No problem, just a hint 😉.