MichaIng/DietPi #2288: Prep-Script on Orange Pi Zero (H2+) WIFI not working

aaro668, 2018-11-26:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=12
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch 9.6
- Kernel version	| `uname -a`
Linux DietPi-test-OPI0 4.14.78-sunxi #412 SMP Fri Oct 26 11:37:04 CEST 2018 armv7l GNU/Linux
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
OrangePi Zero (armv7l)
- Power supply used	| (EG: 5V 1A RAVpower)
5 volt 3 amp Buck converter
- SDcard used		| (EG: SanDisk ultra)
SanDisk Edge 8GB

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
N/A
- Was the software title installed freshly or updated/migrated?
N/A
- Can this issue be replicated on a fresh installation of DietPi?
Yes, see below
- `dietpi-bugreport` ID
N/A

#### Steps to reproduce:

1. Install "Armbian_5.65_Orangepizero_Debian_stretch_next_4.14.78.img" to SD Card
2. Boot SD card on OPI0, log in, change root password. Connect to wifi network using nmtui, verify that wifi is working.
3. Run DietPi prep script as per this link: https://github.com/Fourdee/DietPi/issues/1285#issue-280771944
4. Wifi is now not working anymore, no way to connect to wireless network?

#### Expected behaviour:

- Wifi should work after dietPi Prep script, as it was working before....

#### Actual behaviour:

- Wifi is not working, impossible to connect to wireless network after running DietPi prep script

#### Extra details:

- Are there any special steps needed to enable Wifi on Dietpi OPI0? Wifi is verified to be working with fresh armbian install with good connectivity, and as soon as I run DietPi prep script I am unable to connect to WIFI.....

MichaIng, 2018-11-26:
@aaro668 
Thanks for your report.

Jep it's a known issue when running DietPi-PREP with WiFi connection (SSH) and NetworkManager was installed previously. For this reason currently Ethernet connection is required to have the install done reliable.

However to verify this is really your issue:
- Was the WiFi (SSH) connection dropped during DietPi-PREP running?
- Check whether the `network-manager` is installed on ARMbian image.
- In case, connect via SSH over Ethernet or direct attached HDMI + keyboard.
- Purge the `network-manager` package
- Configure WiFi via `/etc/network/interfaces` and `/etc/wpa_supplicant/wpa_supplicant.conf`.
- Then run DietPi-PREP, which should be then as well work via WiFi SSH connection.

The problem is that when purging `network-manager`, all WiFi related settings are removed and WiFi connection dropped.

aaro668, 2018-11-26:
The issue is not with ssh being dropped....I was ssh through Ethernet, but after running prep script I couldn't set up and use Wi-Fi.....

Aaron Martin
Northway Metals Inc.
aaron@northwaymetals.com
Sales: sales@northwaymetals.com
***Please note new email addresses!***

Twilight0, 2018-11-26:
Same here, I prepared image on Orange Pi Zero H2+ two days ago and wifi is "Not Found".

aaro668, 2018-11-26:
A bit more detail here.....
With the Armbian install the Wifi IS working fine, I believe it has Network Manager, as  I can configure wireless network using nmtui command.
The SSH connection was not dropped during DietPi-PREP, as I was connected over ethernet/serial.
After DietPi-PREP is finished and first boot setup is done I cannot get WIFI set up and working.....
Maybe I need to do it manually some way?

aaro668, 2018-11-27:
Another thing that I noticed was that htop interface changes.....
This is how htop looks with fresh install of Armbian:
[image]
And this is how htop looks after I run DietPi-PREP:
[image]
Maybe this is built into the design of Dietpi to minimize processing?

MichaIng, 2018-11-27:
@aaro668 
The default `htop` does not support CPU temps and frequencies. This seems to be a custom version, e.g.:
- https://github.com/avafinger/htop (amd64+arm64 only)
- https://github.com/wbenny/htop (armhf only)

I already forwarded this request (was done earlier already) to the developer of the official `htop`. Of course the request did already exist 😉: https://github.com/hishamhm/htop/issues/163
- The problem is, that there are many different interfaces to read out the CPU temperature and even if you identify fitting files, sometimes the ones show static degrees only (not real CPU temp) and you need to know which different one you need to read out. Already for our limited amount of supported devices and on Debian-only, it is a mess: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-globals#L1945
But in case of an platform independent tool that should work reliable on all platforms/distros, this is somehow a deal breaker.

Open request for CPU frequencies: https://github.com/hishamhm/htop/issues/129
- I guess it is similar then with the temps.

I would also love to have everything in one place, but the current 3rd party solutions are either outdated (base htop version), or work only on limited architectures and then temps/freqs will not be correct on all devices. Also often implementations are crappy or rely on heavy dependencies and additional services running in background.
So I would definitely stick with the official version, also since on DietPi `cpu` provides CPU temp/freq information already.

Perhaps ARMbian customizes their pre-installed htop for each device, since it goes hand in hand with kernel development 🤔.

aaro668, 2018-11-27:
@MichaIng 
Thanks for the htop info, this is not a deal breaker for me, I just thought I would mention it here as I noticed it at the same time as I found the Wifi problem.
Have you checked in about the Wifi yet? Maybe I need to enable Wifi another way?

MichaIng, 2018-11-27:
NB: There is already on open PR for CPU freq, but no progress for a while 😢: https://github.com/hishamhm/htop/pull/592

About WiFi:
- How did you try to configure WiFi? `dietpi-config`?
- Please check if the interface is up and connected to the AP: `ip a` / `ip r`
- If not, check available network cards: `ls -Al /sys/class/net/` or `cat /proc/net/dev`
- Then check which one is configured in `cat /etc/network/interfaces`, if this is the expected network card.
- Check that WiFi credential are correct: `cat /etc/wpa_supplicant/wpa_supplicant.conf` (of course do not paste this or remove SSID + key here 😉)

And did you try to switch from DHCP to static IP or the other way round? Perhaps also DHCP fails somehow.

Twilight0, 2018-11-27:
I tried to manually modprobe its module today:

`sudo modprobe xradio_wlan`

Then I read these things in the dmesg:

 [   12.736501] xradio_wlan mmc1:0001:1: Direct firmware load for xr819/sdd_xr819                                                               .bin failed with error -2
 [   12.736514] xradio_wlan mmc1:0001:1: Falling back to user helper
 [   12.906360] Generic PHY 0.1:01: attached PHY driver [Generic PHY] (mii_bus:phy_addr=0.1:01, irq=POLL)
[   12.909229] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
 [   12.909245] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
 [   12.909705] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
 [   13.370984] thermal thermal_zone0: failed to read out thermal zone (-110)
 [   14.410902] thermal thermal_zone0: failed to read out thermal zone (-110)
 [   14.972280] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
 [   14.972356] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
 [   16.533948] random: crng init done
 [   16.533966] random: 7 urandom warning(s) missed due to ratelimiting
 [   74.731564] xradio_wlan mmc1:0001:1: bh thread exiting

Also these:

`
dietpi@DietPi:~$ cat /proc/net/dev
`
 
Inter- | Receive | Transmit
 face | bytes    packets errs drop fifo frame compressed multicast|bytes    packets errs drop fifo colls 
 carrier compressed
  eth0:   32022     268    0    0    0     0          0         0    78478     234    0    0    0     0       0          0
  lo:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
`

aaro668, 2018-11-27:
@MichaIng 
I use dietpi-config, then "Network Options: Adapters"
[image]
Then I select "Change Wireless Network Settings"
[image]
Then I select OK
[image]
After I exit from dietpi-config now, this is the terminal history:
[image]
Maybe this can help you determine what I am doing wrong?

aaro668, 2018-11-27:
Here is output from ls /lib/firmware with fresh Armbian install:
[image]
And here is the output after running DietPi-PREP : 
[image]

It looks like there were a lot added, but the xr819 is missing.....

aaro668, 2018-11-27:
So, copying the xr819 directory into /lib/firmware/ directory I am now able to connect to an SSID.
Is this a bug in DietPi-PREP?

Fourdee, 2018-11-28:
@aaro668 

> Is this a bug in DietPi-PREP?

Custom firmware provided by the ```armbian-firmware``` package conflicts with default Debian packages, and, lacks the custom firmware.

I'll fix PREP to detect ARMbian and prevent firmware removal/changes.

aaro668, 2018-11-28:
@Fourdee 
Thanks, let me know when fix is ready to test, then I will go through and make sure it is working here.

Fourdee, 2018-11-28:
@aaro668 

Thanks 👍 

Ok code done, available via dev branch, use below PREP:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```

Fourdee, 2018-11-28:
@aaro668

> Which of these 3 options should I select?

Master should be fine, updated PREP script is already loaded.

Fourdee, 2018-11-30:
This is resolved in PREP ```dev``` and by selecting install branch ```dev``` only. Due to master branch pre-req firmware during wifi set.

Completed, passed local testing.

closed 2018-11-30 by Fourdee