MichaIng/DietPi #2289: alsa-base missing

OE333, 2018-11-26:
Hello,
I just upgraded to 6.18.12 (beta) and found my USB DAC not working properly any more.

My findings so far:
dpkg -l  shows that alsa-base seems not to be installed (but alsa-utils is installed).

Trying to install alsa via dietpi-software does not change the situation.

Trying to install alsa-base via 'apt-get install alsa-base'  brings error message: "Package alsa-base is not available.... "

So the alsa-base package seems to be missing.

#### Required Information:
- DietPi version	= 6.18.12 (beta)
- Distro version	= 9.6
- Kernel version = 4.14.0
- SBC device     = NanoPi NEO 2 (aarch64)

Thanks for any help,
OE333

Fourdee, 2018-11-26:
@OE333 

Hi,

If you updated when the DAC was unplugged/powered off, the detection may have failed and reset the soundcard on system.

```alsa-utils``` is the only package required, contains the alsa libs

Try re-selecting USB DAC in ```dietpi-config``` > Audio options. Reboot system and check sound is functional.

If problems persist, please send us a bug report
```
dietpi-bugreport
```

Fourdee, 2018-12-10:
No response. Please reopen if required.

closed 2018-12-10 by Fourdee

adminy, 2020-08-19:
ok, so I don't think this is related much but my bass and tremble controls are missing also with usb dac, any idea why this might be?