MichaIng/DietPi #2296: Tinkerboard Chromium not working

Stogoh, 2018-11-27:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -a`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

MichaIng, 2018-11-29:
@Stogoh
Thanks for your report. We will investigate.

Stogoh, 2018-11-29:
@MichaIng 
**Sidenote**
I had the same problem with TinkerOS **after sudo apt-upgrade**. But when I just use the image which is online and **not** update it, then chromium is working. Maybe some package is getting updated and then does not work anymore.

MichaIng, 2018-11-29:
Thanks for the info. So we need to check which version still work, what changed and what we need to adjust e.g. in xorg.conf to get it running again.

Fourdee, 2018-11-29:
@Stogoh 

Looks like an issue with GPU accell, as a workaround for now, try disabling the glamor driver:
```
mv /etc/X11/xorg.conf.d/20-modesetting.conf /etc/X11/xorg.conf.d/20-modesetting.conf.disable
```

Stogoh, 2018-11-29:
I have now disabled the driver, but now it fails to initialize xserver. So I‘m suck in the console right now.

Fourdee, 2018-11-29:
@Stogoh 

Updating our TinkerOS image, i'll take a look in the next hour or two.

Fourdee, 2018-11-30:
Confirmed issue. 

@Stogoh 
To return X back to working state:
```
mv /etc/X11/xorg.conf.d/20-modesetting.conf* /etc/X11/xorg.conf.d/20-modesetting.conf
```

Fourdee, 2018-11-30:
Still occurs with v67...
```
wget https://snapshot.debian.org/archive/debian-security/20180701T015633Z/pool/updates/main/c/chromium-browser/chromium_67.0.3396.87-1~deb9u1_armhf.deb
dpkg -i *.deb
rm *.deb
```

Same with egl flag removed.

Fourdee, 2018-11-30:
Note to self:
- Image updated to mainline
- midgard driver with EGL support
- Kodi support
- Do not release until v6.19 is live, due to dev prep'ed and desktop changes

ref: https://github.com/Fourdee/DietPi/commit/fc823eb63c6797cc6fd99ed88d9f6ca01158070d

Stogoh, 2018-12-06:
Well. I just updated to the newest version of DietPi and the problem returns again.

MichaIng, 2019-01-05:
@Fourdee 
The new images are in place, if I see right? If still an issue with current Chromium, is there something we can do about running systems? Any chance to update them to mainline kernel on `dietpi-update`? I guess too risky and might break custom setups that depend on current kernel. So at least should be only optional...

MichaIng, 2019-03-02:
I guess the issue has been resolved with the new images. I mark this as closed for now, but feel free to reopen if required.

If this is really kernel related, I see no chance that we can automate a fix via update. This has to be done manually to assure no unexpected issues occur, or, at best flash the new image.

hekigan, 2019-04-26:
Hi!
I see that this thread is closed but I encounter the same issue.
Yesterday I received the Tinker Board S and did a fresh install.

From LXDE I  start Chromium, it shows and closes back right away...

Just as an extra info, when installing, I got a 404 when trying to get some data from the Fourdee repo. Not sure if could be related in any way though.

MichaIng, 2019-04-27:
@hekigan 
> Just as an extra info, when installing, I got a 404 when trying to get some data from the Fourdee repo. Not sure if could be related in any way though.

I just went through the things we download from dietpi.com and one thing is: https://dietpi.com/downloads/binaries/asus/ASUSTB_GPU.7z
If this failed, then I expect Chromium fails. On the other hand LXDE fails then as well as whole Xserver fails.
Can you please paste: `dpkg -l xserver-xorg-core`

Ah and did you install LDXE independently before or both in the same DietPi-Software run?

hekigan, 2019-04-27:
I did the install as a fresh install and both on the same run.
Since then I used the tip above to install an older version of chromium to make it work.

Below the result of your query:

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
hi  xserver-xorg-c 2:1.19.3-1   armhf        Xorg X server - core server
```

MichaIng, 2019-05-01:
@hekigan 
Okay actually I believe this has not yet finally resolved, although @Fourdee said
> Confirmed Chromium under autostart support

The `xserver-xorg-core` is at least what it should be, so ASUSTB_GPU.7z above got downloaded and installed successfully.

I will reopen the issue now. Which version of Chromium does work for you?

MichaIng, 2019-06-08:
@Stogoh 
Could you please try out the following:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/asus/testing/ASUSTB_GPU.7z
7zr x ASUSTB_GPU.7z
rm ASUSTB_GPU.7z
apt-mark unhold xserver-common xserver-xorg-core
dpkg -i *.deb
G_AGF
apt-mark hold xserver-common xserver-xorg-core
```

Stogoh, 2019-06-08:
> @Stogoh
> Could you please try out the following:
> 
> ```
> cd /tmp
> wget https://dietpi.com/downloads/binaries/asus/testing/ASUSTB_GPU.7z
> 7zr x ASUSTB_GPU.7z
> rm ASUSTB_GPU.7z
> apt-mark unhold xserver-common xserver-xorg-core
> dpkg -i *.deb
> G_AGF
> apt-mark hold xserver-common xserver-xorg-core
> ```

@MichaIng 
I've done it. Chromium is still working.

BTW: I had to enter `G_SUDO` before executing `G_AGF`

MichaIng, 2019-06-08:
@Stogoh 
> I've done it. Chromium is still working.

Good so far, would be great if you could test it with the new(est) Chromium version.

> BTW: I had to enter G_SUDO before executing G_AGF

Jep expected of course 🙂.

Stogoh, 2019-06-08:
Well. I updated chromium from version `67.0.3396.87` to `72.0.3626.122` and it's not working again. Chromium just flashes white for a bit. I've also executed G_AGF command again. Still not working.

MichaIng, 2019-06-08:
@Stogoh 
To verify image and kernel version:
```
cat /etc/armbian_version
uname -a
```

Stogoh, 2019-06-08:
### "Uname -a" outputs:
`Linux Barfritz 4.4.103+ #1 SMP Fri Jun 22 16:14:49 CST 2018 armv7l GNU/Linux`

MichaIng, 2019-06-09:
@Stogoh 
Ah it looks like you use an old TinkerOS based image. The new one is based on ARMbian which ships a new mainline based kernel, which might have the required support, not sure. Sadly AFAIK it is not possible or at least difficult to migrate to ARMbian kernel + bootloader basis, so either you need to reflash or of course stay with the older Chromium version, which should be not that bad.

But I will keep the issue open to see if other ATB users with the new image run into the issue as well.

Thanks for testing the new graphics+xorg packages btw. which do not solve the issue sadly but do not hurt as well and ship a slightly enhanced media feature support + newer library versions.

MichaIng, 2019-09-03:
I mark this issue as closed. New Buster image is up. Tester required for GPU acceleration support: https://github.com/MichaIng/DietPi/issues/3094

closed 2019-09-03 by MichaIng