MichaIng/DietPi #2298: Chromium - Dedicated use, without desktop : No such file or directory

davidhq, 2018-11-28:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`

#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=18
G_DIETPI_VERSION_RC=14
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

9.6

- Kernel version	| `uname -a`

Linux DietPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)

RPi 3 Model B (armv7l)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

YES

- `dietpi-bugreport` ID

ce5fdb5b-eab0-4c43-aab5-03dfac604491

#### Steps to reproduce:

1. ... Fresh install of dietpi
2. ... go to software-install and select Chromium - Dedicated use, without desktop

When system boots, it's still in terminal saying:

/DietPi/dietpi/login: line 125: /var/lib/dietpi/dietpi-software/installed/chrome-autostart.sh: **No such file or directory**

#### Expected behaviour:

- I believe Chromein Kiosk mode with specified url open should appear!

#### Actual behaviour:

- "Nothing too promising"

Fourdee, 2018-11-28:
@davidhq 

Many thanks for the report 👍 

I'll try to replicate.

Fourdee, 2018-11-28:
@davidhq 

Just a quick question, have you installed Chromium in ```dietpi-software```? Although you can select the autostart option, it will not install the software, that needs to be done in ```dietpi-software```.

davidhq, 2018-11-29:
Hi! You are right, sorry, didn't realize this... maybe you could add a small comment that this has to be done? I mean, it is obvious but maybe someone (like me) would assume this is already in base image and you just set it to autostart...

So from this perspective the issue is solved but if you don't mind I'd like to add a small question: everything works except I cannot make use of full horizontal resolution from self-standing Chromium on DietPi... I tried everything... I'm using Waveshare 10.1" 1280×800 HDMI touchscreen... works fine on Raspbian with GUI... here on DietPi when it boots I can also see some text extended accoss the full width...

Here is the issue: 
[image]

I tried everything I could by editing /DietPi/config.txt and /DietPi/dietpi.txt

in later there is no much to do:

```
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=800
```

in the former I tried every possible combination of these settings (including uncommenting):

```
disable_overscan=1
framebuffer_width=1280
framebuffer_height=800
```
and settings from waveshare support document

namely:

```
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1280 800 60 6 0 0 0
hdmi_drive=1
```

```
max_usb_current=1
hdmi_group=2
hdmi_mode=27
hdmi_drive=1
```

I'm starting Chromium with:

    CHROMIUM_OPTS='--kiosk --test-type --incognito --start-maximized --disable-pinch --overscroll-history-navigation=0 --window-size=$RES_X,$RES_Y --app-window-size=$RES_X,$RES_Y'

Thank you for any tips...

MichaIng, 2018-11-29:
Good point about autostart vs installed software.
Where a certain software install is required, we should check for install status and in case inform user, offering to run the install directly via whiptail menu.

Fourdee, 2018-11-29:
@davidhq 

Please try editing:
```
/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
```

Change the following line to below:
```
CHROMIUM_OPTS='--kiosk --test-type --incognito --start-maximized --disable-pinch --overscroll-history-navigation=0 --window-size=$RES_X,$RES_Y --start-fullscreen'
```

Save changes and reboot, see if that resolves the issue.

Fourdee, 2018-11-29:
@davidhq 

If you could confirm the follow change works on your system, i'd be grateful:
```
https://github.com/Fourdee/DietPi/issues/2298#issuecomment-442995196
```

Fourdee, 2018-11-30:
@davidhq 
Found the issue, was able to replicate with ASUS TB, solution is to copy and paste all the below code into terminal:
```
cat << _EOF_ > /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
#!/bin/bash
#Autostart run script for Kiosk mode, based on @AYapejian https://github.com/Fourdee/DietPi/issues/1737#issue-318697621
# - Please see $HOME/.chromium-browser.init (and /etc/chromium.d/custom_flags) for additional egl/gl init options

# Command line switches https://peter.sh/experiments/chromium-command-line-switches/
# 	--test-type gets rid of some of the chromium warnings that you may or may not care about in kiosk on a LAN
#   --pull-to-refresh=1
#   --ash-host-window-bounds="400,300"

# - Resolution to use for kiosk mode, should ideally match current system resolution
RES_X=\$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_RES_X=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')
RES_Y=\$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_RES_Y=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')

CHROMIUM_OPTS="--kiosk --test-type --window-size=\$RES_X,\$RES_Y --start-fullscreen --start-maximized --window-position=0,0"
# - If you want tablet mode, uncomment the next line.
#CHROMIUM_OPTS+=' --force-tablet-mode --tablet-ui'

# - Add URL for first run:
URL=\$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')
CHROMIUM_OPTS+=" --homepage \$URL"

#Find absolute filepath location of Chromium binary.
FP_CHROMIUM=\$(which chromium)
if [[ ! \$FP_CHROMIUM ]]; then

	# - Assume RPi
	FP_CHROMIUM="\$(which chromium-browser)"

fi

xinit \$FP_CHROMIUM \$CHROMIUM_OPTS
_EOF_
```

davidhq, 2018-11-30:
testing right now, first run wasn't successful, when I replaced the new autostart file, chromium didn't start, device stayed in terminal... I'll try to find the reason and let you know

davidhq, 2018-11-30:
it actually did work, I made a mistake... so, for now it looks much better... will test some more if there is any other problems, otherwise I think this resolves the issue....

davidhq, 2018-12-01:
OK, thank you.

Regarding unclutter: I didn't want to use that since I have experience with this better solution as mentioned:

https://raspberrypi.stackexchange.com/questions/53127/how-to-permanently-hide-mouse-pointer-or-cursor-on-raspberry-pi

Can you please see where and how this can be achieved here, then you could add option for that...

davidhq, 2019-01-29:
@Fourdee are there maybe any new hints on how to hide the cursor without "polling"? 

thank you

davidhq, 2019-02-09:
I resolved the issue by going back to official Raspbian Desktop release... 

I was very surprised when I started booting two equivalent tablets side by side and official release was quite a few seconds faster to display my site contents. Also the black screen with nothing happening (which could confuse users) was shown for much less time.

So I guess this is not a project I'm very interested in anymore but thank you for teaching me a few things. Good luck! 

Could also be true that when you try to make things easier for a lot of people, you fall on your face and nothing really works in the project anymore... don't know, just guessing... anyway good luck again.

MichaIng, 2019-02-09:
@davidhq 
Might I ask how you installed Chromium kiosk mode on Raspbian? By comparing the start scripts we might be able to identify conflicts with the `-nocursor` option.

davidhq, 2019-02-11:
Also I noticed something quite strange on my last tests before I purge DietPi from all raspberries... background image (jpg) on my sites were pixelated... I was worried that one of the tablet screens is worse quality but thankfully it was only something screwed up in dietpi (again)... also text size was different, I was wondering why different font size on official raspbian and dietpi... again dietpi thing was "out of the ordinary"... I wish you all the best with project and "I'm outta here" ;) did learn a few interesting things though - thank you