MichaIng/DietPi #2300: Python script at startup (dietpi config) will never let you abort it

joyeriastv, 2018-11-29:
## Creating a bug report/issue:

I am not sure if this is a bug, but  dietpi soft should have an option like boot up types before running that script at startup to revert it, does it have?

I have a python script running without login at startup, created with the dietpi options (last one) to run anything at startup, now I cant even close it with Ctrl+C, how can I revert it without using a second linux PC to edit files?

thanks in advance

MichaIng, 2018-11-29:
@joyeriastv
You man you use dietpi-postboot with /var/lib/dietpi/postboot.d/* to start your script?

Since postboot is started via systemd unit, is is executed non-interactively outside of terminal. You should only use this for oneshot scripts that themself exit. If you want to start a script that stay running interactively, you should do start on login, e.g. via: /etc/profile.d/*.sh
I'm not aware of a method to run a script interactively on terminal, but before actually login.

Another possibility is to create an own systemd unit to start at desired boot stage. It will be always non-interactive but you are able to stop it via: `systemd stop servive_name`

MichaIng, 2018-11-29:
Okay, just read the issue on the forum (I guess this is yours as well?): https://dietpi.com/phpbb/viewtopic.php?f=11&t=5285

I understood now that it is not only a question but urgent help request 😉.
If you added the script to autostart via `dietpi-config`, then it must be `dietpi-autostart`, which starts the script via systemd unit prior/regardless of login.

To stop it: `systemctl stop dietpi-autostart_custom` 😄 
Generally you can use `htop` to kill processes as well. The above is the "soft" way for systemd services.

MichaIng, 2018-12-09:
Will mark as closed, since question has been answered/issue solved. Feel free to reopen, if required.

closed 2018-12-09 by MichaIng