MichaIng/DietPi #2301: RPI SBC | RPi 3 Model A+

adamotte, 2018-11-29:
## Creating an image request:

#### Give us some formal device information:
- Device name | **Raspberry Pi 3 Model A+**
- Official product URL | https://www.raspberrypi.org/products/raspberry-pi-3-model-a-plus/
- GitHub resource URL | https://github.com/raspberrypi
- Image download URL | https://www.raspberrypi.org/downloads/

MichaIng, 2018-11-29:
@adamotte
Thanks for your request. Actually the RPi image should work for all RPi versions. Did you face any issues with it?

Fourdee, 2018-11-30:
DietPi naming for 3A+ will be fixed in v6.19. 

@adamotte 
Image should work out the box with 3A+. However, if you can confirm, i'd be grateful.

adamotte, 2018-11-30:
My ticket was just to keep in mind this new device and double check if, has you said, it works out of the box.
He is not available yet in my country, i will buy it as soon as i can.

Fourdee, 2018-12-01:
@adamotte 

> My ticket was just to keep in mind this new device and double check if, has you said, it works out of the box.
He is not available yet in my country, i will buy it as soon as i can.

Yep, thank you for the heads up 👍 

I'll order one when I can and verify. Unless someone else can confirm its fine.

Fourdee, 2019-01-28:
Results now in survey, will mark as completed.
```

RPi 3 Model A+ (armv7l) | 23
-- | --

```

closed 2019-01-28 by Fourdee