MichaIng/DietPi #2302: DietPi-Config | Change the number of WiFi Slots in Wireless Network Options

craviee, 2018-11-29:
It is possible to change the number of WiFi Slots of 5 to 10 or 20?

I tried to change the variable MAX_SSID_SLOTS of /boot/dietpi/func/dietpi-wifidb to 10 but nothing happened, and when I reboot the system the variable returns to the default 5.

MichaIng, 2018-11-29:
@craviee 
You need to adjust the value within `/DietPi/dietpi/func/dietpi-wifidb`. On reboot this will be ported to `/boot/dietpi/func/dietpi-wifidb` as well.

But a dietpi-update will revert of course.
My aim, perhaps can be done for v6.19, is to have no a static max slot amount, but simply start with one slot. Then have a button within `dietpi-wifidb` menu to add an additional slot up to no limit.

craviee, 2018-11-29:
> @craviee
> You need to adjust the value within `/DietPi/dietpi/func/dietpi-wifidb`. On reboot this will be ported to `/boot/dietpi/func/dietpi-wifidb` as well.
>
> But a dietpi-update will revert of [...]

This fully resolve my issue.
May I close this issue?

MichaIng, 2018-11-29:
Thanks for feedback, yes you can close. The feature/improvement will be done via separate issue.

closed 2018-11-29 by craviee