MichaIng/DietPi #2307: DietPi-Update | fails on RPi 3 Model B+

k-plan, 2018-12-01:
Required Information:

- DietPi version	| `cat /DietPi/dietpi/.version`

```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=0
G_GITBRANCH=master
G_GITOWNER=Fourdee
```
- Distro version	| `echo $G_DISTRO_NAME or cat /etc/debian_version`
```
stretch
9.6
```
- Kernel version	| `uname -a`
`Linux-3 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`

- SBC device	| `echo $G_HW_MODEL_DESCRIPTION`
`RPi 3 Model B+ (armv7l)`

- Power supply used	| (EG: 5V 1A RAVpower) 
`RaspBerry Pi 3 Official power supply 5.1V / 2.5A`

- SDcard used	| (EG: SanDisk ultra)
`Toshiba EXCERIA PRO R95 16GB Silver`
[image]

Steps to reproduce:

- Download and write last DietPi image to sd card: `DietPi_v6.17_RPi-ARMv6-Stretch.img`

```
:~# dietpi-update 2

[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.17.0
[ INFO ] DietPi-Update | Latest version  : v6.18.14
```

```
:~# dietpi-update 1

[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.17.0
[ INFO ] DietPi-Update | Latest version  : v6.18.14
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=12359 MB | required=100 MB
[  OK  ] DietPi-Update | Update is being applied, please wait...
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
[ INFO ] Applying critical pre-patches

 DietPi-Pre-patch
─────────────────────────────────────────────────────
[ INFO ] Completed pre-patches with exit code: 0

[  OK  ] DietPi-Update | APT update, please wait...
Hit:1 https://mirror.netcologne.de/raspbian/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...

[  OK  ] DietPi-Update | G_AGUP
[  OK  ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Update | G_AGUG
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119    0   119    0     0    191      0 --:--:-- --:--:-- --:--:--   191
curl: (7) Failed to connect to codeload.github.com port 443: Connection refused
[FAILED] DietPi-Update | Download failed, unable to run update. Please try running dietpi-update again.
```

Fourdee, 2018-12-02:
@k-plan 

Hello you, long time no see 👍 Hope all is well?

Strange, working fine here, could be a DNS issue, what happens if you do:
```
wget https://github.com/Fourdee/DietPi/archive/master.zip
curl -kL https://github.com/Fourdee/DietPi/archive/master.zip > test
```

Whats your DNS set to, Pihole installed?

k-plan, 2018-12-02:
@Fourdee 

_Hey Dan, more or less at the moment.  Hope all is well on your side._

Thanks for your reply. 

Yes, piholed. Yes, redirection to `https://codeload.github.com/Fourdee/DietPi/zip/master` is blocked by : 
```
Match found in https://urlhaus.abuse.ch/downloads/text/:
   codeload.github.com
```

But I don't understand, why `dietpi-update` then didn't use the mirror:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L91-L94

Fourdee, 2018-12-03:
@k-plan 

Strange, no block with default Pi-hole install in my testing, custom additions?

> But I don't understand, why dietpi-update then didn't use the mirror:

Try disabling the github entries ```#``` in ```dietpi-update```
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L77-L103

Then re-run the update. That will switch to DietPi mirror.

k-plan, 2018-12-03:
@Fourdee 

> Strange, no block with default Pi-hole install in my testing, custom additions?

Yes, see above.

> `https://urlhaus.abuse.ch/downloads/text/`

Add https://urlhaus.abuse.ch/downloads/text/ to your pihole blocklist. 
About this list: https://urlhaus.abuse.ch/

> Try disabling the github entries `#` in `dietpi-update`
> Then re-run the update. That will switch to DietPi mirror.

Yes, indeed! Comment [line 93](https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L93) out  suffice.
But why to have a mirror, if it will never be used? 😉

Fourdee, 2018-12-03:
@k-plan 

> But why to have a mirror, if it will never be used? 

Strange, should fail and switch to the backup mirror:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L107-L112

It must be able to download https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L79, but not the ZIP.

k-plan, 2018-12-03:
@Fourdee 

> Strange, should fail and switch to the backup mirror:

No, it will not fail. `https://raw.githubusercontent.com/ ....` is reachable.
But `https://github.com/Fourdee/DietPi/archive/master.zip` is a redirecton to `https://codeload.github.com/Fourdee/DietPi/zip/master`
and `https://codeload.github.com/....` will be blocked by pihole. So:

```
[  OK  ] DietPi-Update | G_AGUG
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119    0   119    0     0    191      0 --:--:-- --:--:-- --:--:--   191
curl: (7) Failed to connect to **codeload.github.com** port 443: Connection refused
[FAILED] DietPi-Update | Download failed, unable to run update. Please try running dietpi-update again.
```

Fourdee, 2018-12-03:
@k-plan 

Not much we can do about Github's redirect. 

We could add an option in ```dietpi-update``` to force selecting between Github and DietPi mirror?

k-plan, 2018-12-04:
@Fourdee 

`Unable to replicate ☕️`

Really? I have explain it twice what is to do and it persistent exist on all my devices not only RPI3B+ I have test it at last.

Sorry, last trying now.

```
DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.18.14
[ INFO ] DietPi-Update | Latest version  : v6.19.5
```
The problem is your `DietPi-Update | Checking mirror` =>  `DietPi-Update | Using update server`

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L107-L128

You have to check reachability of `DietPi-master.zip` as well: `DietPi-Update | Checking mirror`

```
:~# curl -kL https://github.com/Fourdee/DietPi/archive/master.zip > test
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119    0   119    0     0    192      0 --:--:-- --:--:-- --:--:--   192
curl: (7) Failed to connect to codeload.github.com port 443: Connection refused
```

```
:~# curl -kL https://dietpi.com/downloads/dietpi-update_mirror/master/DietPi-master.zip > test
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2474k  100 2474k    0     0  3644k      0 --:--:-- --:--:-- --:--:-- 3644k
```

and then select a update mirror `DietPi-Update | Using update server` 

But anyway, I don't care.  Do what ever you like. I can not do more then point to a possible problem. For me it's solved.

closed 2018-12-04 by k-plan