MichaIng/DietPi #2308: DietPi-Software | Pydio: Pretty URL rewrite required for sync clients

MichaIng, 2018-12-01:
https://dietpi.com/phpbb/viewtopic.php?f=11&t=484&p=15596&sid=79a48e972c5bbda81778deff5b9ef565#p15596

MichaIng, 2018-12-02:
@Fourdee
I will add this to v6.19 now:
- Lighttpd according to solution in forum. Hope user can verify it's working there (with Windows client). I tested with another DietPi VM as client. Connection worked (without not), but last step to start dir sync somehow didn't start it. Pressing the button didn't have an effect. Not sure if something is missing for the client binary, since I just installed required libraries according to execution error messages. Didn't find a list about requirements and there is no installer available for the linux client, just the bare binary.
- Apache works out of the box with .htaccess
- Nginx still needs check, but should be fast forward with docs and similar to ownCloud/Nextcloud config.

MichaIng, 2018-12-02:
Damn, more complicated rewrite rules required, since php files are as well inside subdirs with dots in their path... https://dietpi.com/phpbb/viewtopic.php?f=11&t=484&start=20

MichaIng, 2018-12-02:
php-cli is recommended within docs and also shown on Lighttpd install guide: https://github.com/pydio/pydio-doc-howto/blob/master/system_installing_on_fedora_lighttpd.md#init-2

Used for background jobs, if I understood right.

MichaIng, 2018-12-02:
Just found: https://github.com/Fourdee/DietPi/issues/297
That actually gives a hint about the different required rewrite rules... Why there is no official doc about this, which all seems to be required... 🤔

MichaIng, 2018-12-02:
Gave up in Linux client. Most properly some library is missing.

🈯️ Test with Windows client succeeds with:
```
$HTTP["url"] =~ "^/pydio($|/)" {
        # Deny access to pydio/data dir
        $HTTP["url"] =~ "^/pydio/(data|conf|core)" {
                url.access-deny = ("")
        }

        # Pretty URL rewrite, required for sync clients
        url.rewrite-if-not-file = ( "^/pydio/(.+)/?$" => "/pydio/index.php/$1" )

        # Disable PHP output buffering, recommended by Pydio
        setenv.add-environment += ( "PHP_ADMIN_VALUE" => "output_buffering=Off" )

}
```
- Added more access denials according to `.htaccess` files. Will add again more.

MichaIng, 2018-12-02:
Nginx example conf... to a while to find it 😆: https://github.com/pydio/pydio-core/blob/develop/core/src/nginx.conf.sample

Fourdee, 2018-12-03:
🈯️ Test install Lighttpd , not tested Sync client yet.

Fourdee, 2018-12-04:
Need to verify:
- Sync clients functional with:
-- Lighttpd
-- Nginx
-- Apache2

Then we can mark this as completed.

Fourdee, 2018-12-04:
@MichaIng 

Works a charm, great work 👍

closed 2018-12-04 by Fourdee