MichaIng/DietPi #2310: Nextcloud Talk Install Fail

tenorable, 2018-12-02:
#### Details:
- Date           | Sun  2 Dec 01:13:04 CST 2018
- Bug report     | N/A
- DietPi version | v6.18.14 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | ncc maintenance:mode --off
- Exit code      | 127
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...I have my files saved on an external drive. Copied the dietpi_userdata/nextcloud_data directory to the usb drive. Changed the nextcloud-config.php file to point to the usb drive.  All seems to be working correctly, can upload and access files
2. ...used the dietpi-software to install Nextcloud Talk

#### Expected behaviour:

- ...Talk should be installed with the Turn server

#### Actual behaviour:

- ...failed.  Said there was no ncc command, and there isn't it is occ.

#### Extra details:

- ...If I just add it from within Nextcloud, it seems to work ok.

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1366: ncc: command not found
```

MichaIng, 2018-12-02:
@tenorable
Thanks for your report. Will investigate.

Nextcloud install went fine, or did you already have it installed before?

tenorable, 2018-12-02:
It did not go fine at first because some of the config files from the
previous attempt were there, but after deleting then it went OK. Any advice
on how to get the mobile app to talk with the server? Having trouble with
that. Say malfigured server when the app tries to connect.

Thanks,
Kevin

MichaIng, 2018-12-03:
@tenorable
Do calls from browser to browser work well? Then desktop browser to WebRTC capable mobile browser?
Does either the Nextcloud log or turn server log (/var/log/turn_*.log) give some hint?

Will try with fresh setup as well.

tenorable, 2018-12-03:
So, what I did for the mobile app was uninstall it, deleting the previous
files, and then reinstalled it.  Then it seemed to work OK, but took a long
time to finally get in and sync.

Thanks,
Kevin

MichaIng, 2018-12-04:
@tenorable 
Okay did some local testing here, generally it works. On local network connection is very fast, though outside of local network via TURN server indeed takes longer. This highly depends on how the TURN server (your Nextcloud server) is connected to www, NAT, router and firewall setup.

The job of the TURN server is to navigate traffic through different NAT setups successfully, which is otherwise not always possible natively by P2P WebRTC video calls and such, even with STUN server usage (exchange each users local IP as well, besides the external IP only). But it does not work 100% reliable. I also faced issues when doing intercontinental calls e.g. and on Nextcloud forums there are by times similar reports as well, even with all settings done according to docs and Nextcloud Talk/Spreed devs themselves.
It's gotten already better and I guess will become more stable with increasing coturn and Nextcloud Talk versions, in terms of reliability and performance. Note that WebRTC is still a quite young technology, and also e.g. the browsers did implement this just recently, for sure all with slight differences that need to be handled.

Generally if you face issues, please report them here: https://help.nextcloud.com/t/howto-setup-nextcloud-talk-with-turn-server/30794 We can then forward them to either coturn, Nextcloud Talk web or mobile app, depending on where it belongs to.
- Our implementation is based on this HowTo (mine 👼), which again is based on official docs, discussions with devs and gathering all info/experiences I could.
- There we have a larger audience and devs that we can reach.
- If some new info/improvements about TURN server settings are found, I will immediately add them to our install code.

Only thing which is important to report here on DietPi GitHub/forum is that dietpi-software install process itself. Everything should finish without errors or course.
_______________
About performance as well:
- Please check the notes about logging in the HowTo. By default coturn logs very much to `/var/log/turn_*.log`, which is not required besides for debugging. Sadly you cannot adjust the log level, just the log target (file and/or syslog and/or stout). And even with everything disabled and log file to /dev/null, on service start one `/var/log/turn_*.log` is created to log startup messages, obviously before the logging settings are applied.
- Actually this is something we can improve in our install code:
  - Reference: https://groups.google.com/forum/#!topic/turn-server-project-rfc5766-turn-server/k1VzO8blRiQ
  - By default disable all coturn logging via service.d execution command extension. Otherwise at least every service start, one log file is created.
  - If I understood correct, turnserver.conf settings will, if set, overwrite cmd options. Verify this, to allow user adjust, if required.
  - Add notes about this to our online doc.

closed 2018-12-04 by MichaIng