MichaIng/DietPi #2311: Unstable Internet Connection with Raspberry Pi Zero with 6.18

WTFMaster, 2018-12-02:
Hey Guys,

I am using an Raspberry Pi Zero W and connect it with Wifi. When I keep IPv6 enabled, the Internet connection is very unstable. When I turn IPv6 off - everything works fine.

Is this a known issue?

MichaIng, 2018-12-02:
@WTFMaster 
Thanks for your report.

When you enable IPv6, do you completely disable IPv4 on local network, so the Pi has no IPv4 address assigned at all?

It is known that DietPi currently relies on a local IPv4 network, at least, besides IPv6, there need to be an IPv4 address assigned to the network adapter/interface as well.

This is something we need to work on, to allow using IPv6 network only.

Fourdee, 2018-12-03:
@WTFMaster 

Very strange, IPv6 is having this effect.

After boot, try:
```
iwconfig wlan0 power off
```

See if that improves stability.

Also please check the signal strength in ```dietpi-config```, or, 
```
iwconfig wlan0
```

Fourdee, 2018-12-10:
No response, will mark as closed. Please reopen if required.

closed 2018-12-10 by Fourdee