MichaIng/DietPi #2318: Suspend to RAM not working on Pinebook 1080p (SOPINE A64)

Maldus512, 2018-12-04:
#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=18
G_DIETPI_VERSION_RC=14
G_GITBRANCH=master
G_GITOWNER=Fourdee
```

- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
    `stretch`
- Kernel version	| `uname -a`
     `Linux DietPi 4.19.2-sunxi64 #92 SMP Thu Nov 22 19:26:30 CET 2018 aarch64 GNU/Linux`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
     `Pinebook 1080p (aarch64)`
- Power supply used	| (EG: 5V 1A RAVpower)
     `5V 3A (official Pinebook power supply)`
- SDcard used		| (EG: SanDisk ultra)
     `16GB internal eMMC`

#### Steps to reproduce:

1. Install DietPi on Pinebook; flash the image on an SD card and then dd it on the internal eMMC module, as per instructions here https://github.com/Fourdee/DietPi/issues/2299#issue-385428882
2. run `systemctl suspend` as root

#### Expected behaviour:

- System suspends to RAM

#### Actual behaviour:

- Screen goes black for a while and normal execution resumes. Additionally X server crashes.

#### Extra details:

- dmesg output after attempt:

```
[  632.792505] PM: suspend entry (s2idle)
[  632.792515] PM: Syncing filesystems ... done.
[  632.962214] Freezing user space processes ... (elapsed 0.002 seconds) done.
[  632.964271] OOM killer disabled.
[  632.964273] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  632.965687] Suspending console(s) (use no_console_suspend to debug)
[  632.981420] RTL871X:  suspend start
[  632.981456] RTL871X: rtw_cmd_thread(wlan0) stop_req:1, break
[  632.982027] RTL871X: rtw_dev_unload: driver not in IPS
[  632.988747] RTL871X: rtw suspend success in 4 ms
[  632.988770] dpm_run_callback(): pm_generic_suspend+0x0/0x40 returns -38
[  632.988780] PM: Device mmc1:0001:1 failed to suspend async: error -38
[  633.060889] PM: Some devices failed to suspend, or early wake event detected
[  633.162077] OOM killer enabled.
[  633.162082] Restarting tasks ... done.
[  633.163973] PM: suspend exit
[  633.164189] PM: suspend entry (s2idle)
[  633.164192] PM: Syncing filesystems ... done.
[  633.193336] r8152 1-1:1.0 eth0: carrier on
[  633.346619] Freezing user space processes ... (elapsed 0.001 seconds) done.
[  633.348601] OOM killer disabled.
[  633.348603] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  633.350026] Suspending console(s) (use no_console_suspend to debug)
[  636.356700] dpm_run_callback(): pm_generic_suspend+0x0/0x40 returns -38
[  636.356714] PM: Device mmc1:0001:1 failed to suspend async: error -38
[  636.356744] PM: Some devices failed to suspend, or early wake event detected
[  636.477527] r8152 1-1:1.0 eth0: carrier on
[  636.505808] OOM killer enabled.
[  636.505812] Restarting tasks ... done.
[  636.550309] PM: suspend exit
```

Fourdee, 2018-12-04:
@Maldus512 

Hi, 

Many thanks for the report 👍 

Known issue (apologies forgot to list it), should be resolved with 4.20 kernel. We need to wait :)

Maldus512, 2018-12-04:
@Fourdee 
Hi,
I suspected it, but couldn't find any reference. I'll gladly wait for the next release, thank you.

MichaIng, 2019-06-25:
@Maldus512 
Has been a while. Meantime Fourdee created a new image with much newer kernel (4.19) on ARMbian basis. I hope suspend to RAM works well with this.

MichaIng, 2020-03-13:
Another new image based on Debian Buster is ready for testing: https://github.com/MichaIng/DietPi/issues/2979

I mark this issue as closed due to missing reply. Feel free to reopen it is still present with Buster beta image.

closed 2020-03-13 by MichaIng