MichaIng/DietPi #2319: Initial install fail - Odroid HC1

rlamans, 2018-12-04:
## Creating a bug report/issue:
#### Details:
- Date           | Mon 10 Sep 07:09:19 EDT 2018
- Bug report     | N/A
- DietPi version | v6.14
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | Connection test: https://deb.debian.org/debian/
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-09-10 07:09:17--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘deb.debian.org’
```

rlamans, 2018-12-05:
Yep - initial install problem - self imposed. I think I had a problem because I edited the Dietpi file. Reloaded the op sys, edited the Dietpi file (paying close attention to detail) and all is well. I had a comma between two DNS servers and noticed it when editing Dietpi the second time.

Thanks for the quick response ...

closed 2018-12-05 by rlamans

MichaIng, 2018-12-05:
@rlamans 
Thanks for reply. Had not in mind that indeed you can actively choose the DNS with static IP via `dietpi.txt`. Great it worked with syntax correction.