MichaIng/DietPi #2320: DietPi-Software | Steam Link now in BETA on Raspberry Pi

GulyFMG, 2018-12-04:
Creating a software request:
Give us some formal software information:

    Software title | Steam Link now in BETA on Raspberry Pi 
    Short description | Steam Link now in BETA on Raspberry Pi 
    GitHub URL (if available) | 
    (Official) install documentation | https://steamcommunity.com/app/353380/discussions/0/1743353164093954254/

Are there similar/alternative software titles available with DietPi-Software?

No.

How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

    Some builds don't have a desktop environment so could be used like kodi or some other software and auto start on steam link

Can you provide the installation steps that you would suggest DietPi-Software to do?

curl -#Of http://media.steampowered.com/steamlink/rpi/steamlink_1.0.4_armhf.deb
sudo dpkg -i steamlink_1.0.4_armhf.deb

Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic wink.

Yes

MichaIng, 2018-12-04:
@GulyFMG 
Thanks for your request. We'll see if we can implement. However there are some other requests in the pipe that I personally would give more priority.

Help, e.g. via PR for implementation highly welcome as always 😉.

GulyFMG, 2018-12-04:
No problem, since its a beta release. But thank you for considering it.

MichaIng, 2018-12-26:
Official URL: https://store.steampowered.com/steamlink/about/
Vote for it on FeatHub: https://feathub.com/Fourdee/DietPi/+22

FaserF, 2018-12-30:
Would love to see this! And btw it is out of Beta already :)

enolp, 2019-06-20:
Is pretty simple to install Steam Link on DietPi. All you have to do is to run "apt install steamlink" and then "runuser -l dietpi steamlink". This 2 commads were what I need by having Kodi already installed.

MichaIng, 2019-06-20:
@enolp 
Thanks for sharing. I am currently working on Mycroft AI voice assistant, but especially if that easy, Steam Link should make it into v6.26.

enolp, 2019-06-21:
The Steam Link app itself runs great even from a non-graphical environment (terminal) but there is some library missing when it connects to a computer because it fails. Here you can see the same exact behavior: https://www.reddit.com/r/archlinux/comments/bnracf/steam_link_in_house_streaming_from_my_arch_linux/

I've installed Raspbian (full edition) and it works very (x10) well, using Raspbian Lite i've got the same issue as above. I don't know what is missing and what is bloat

MichaIng, 2019-06-21:
@enolp 
Thanks for testing, we will figure this out.

spschoen, 2020-08-19:
Sorry to necro, just leaving some notes from my installation for anyone that wants to put in a PR for this (which I'll try to do it this weekend, after reading up a bit).

Before installing `steamlink` I had installed `xfce4`, but otherwise a default installation of `dietpi`.  Only two major stumbling blocks to note:

* `steamlink` will install without fanfare but requires `net-tools` and `zenity` to work actually (I think `net-tools` might be skippable but I prefer removing the cause of warnings/errors rather than ignore them)
* dietpi/logged-in user will need to be added to `video`, `plugdev`, `input` groups (I think `video` might be the only required and `steamlink` will add the user to `plugdev` and `input` groups)

Otherwise, it's quite simple to install and get running.  As I said above, I'll take a look at how y'all do PRs and see if I can't get this added to the big list this weekend or next.

Have a good one y'all.

Joulinar, 2020-08-19:
I did something similar couple of days ago

https://dietpi.com/phpbb/viewtopic.php?p=26640#p26640