MichaIng/DietPi #2325: Nginx Reverse proxy is not working after dietpi update

studioai, 2018-12-05:
## Creating a bug report/issue:
I just updated DietPi to 6.19.5 from previous version, then Nginx Reverse proxy stopped working.
other .conf in etc/nginx/seites-deitpi which is not using Reverse proxy works well.

Here's one of my reverse proxy .conf

`location /git/
{
        proxy_pass    http://localhost:3000/;
        proxy_set_header    Host    $host;
        proxy_set_header    X-Real-IP    $remote_addr;
        proxy_set_header    X-Forwarded-Host       $host;
        proxy_set_header    X-Forwarded-Server    $host;
        proxy_set_header    X-Forwarded-For    $proxy_add_x_forwarded_for;
}
`

Restoring my dietpi backup makes reverse proxy work again, I really wonder what is changed and causing the issue in newest version. 

P.S: @Fourdee I really appreciate for your works!

#### Required Information:
- DietPi version	| 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=5
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	| stretch
- Kernel version	| Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux
- SBC device		| Native PC (x86_64)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| Nginx
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

MichaIng, 2018-12-05:
@studioai 
I just checked what we patched with the update. Although we changed some config file names, we did not patch content or something and nothing should affect proxy handling. Only thing I can imagine is that nginx APT package update changed something about the syntax? Do you remember if the nginx package was updated during dietpi-update as well? You can check `cat /var/tmp/dietpi/dietpi-update.log` for this.

But I recognized another issue, perhaps even related to your issue:
- On Nginx we changed our custom sites/config file names inside /etc/nginx/sites-dietpi/ from *.config to *.conf endings, for consistency. But due to back and forth about where these files are placed, I missed to add changing the include directive in `/etc/nginx/sites-available/default` accordingly.
The line at the end need to be `include /etc/nginx/sites-dietpi/*.conf` now instead of `*.config`
- So the configurations from there are not loaded anymore after the patch. Is yours placed there as well?

To fix:
`grep -q 'include /etc/nginx/sites-dietpi/\*\.config;' /etc/nginx/sites-available/default && G_CONFIG_INJECT 'include /etc/nginx/sites-dietpi/\*\.config;' 'include /etc/nginx/sites-dietpi/*.conf;' /etc/nginx/sites-available/default`

@Fourdee 
Damn mistake, my fault. Should have shifted the internal config naming to v6.20 with full beta phase and such, since this was not required for changing the names of those files in `.conf/dps_*`.

Not sure how to handle this best now. Pydio, Nextcloud, ownCloud, rutorrent and some other should be broken (or missing settings/features) on Nginx after v6.19 patch 🤔.

studioai, 2018-12-06:
@MichaIng 
Thanks for your reply.
I remember there was no nginx update during Dierpi update process for sure.

There's a conflict on config files for pydio because I have redirection config for pydio already and 6.19.5 adds a config for pydio automatically.  So, I just delete my old one. and It works again.

Fourdee, 2019-01-13:
@MichaIng 

We could try linking the below in changelog:
https://github.com/Fourdee/DietPi/tree/master/.conf

Then users can check version/modified date to see if any files have been replaced/updated.

---

Another option is to add tweak ```Download_Install``` a bit.

Fourdee, 2019-01-13:
Went with ```G_WHIP_MSG```, if it becomes too "annoying" for end users, we can change to ```G_DIETPI-NOTIFY 2```.

Fourdee, 2019-01-13:
Yep working fine:
[image]

Completed.

closed 2019-01-13 by Fourdee