MichaIng/DietPi #2328: Jackett will not load

19eighties, 2018-12-06:
## Creating a bug report/issue:

Jackett no longer launches after update to 6.18 and 6.19.

#### Required Information:
- DietPi version	| v6.19.5
- Distro version	| stretch
- Kernel version	| Linux DietPi 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux
- SBC device		| Rock64 (aarch64) 
Power supply used	| 12V 3A 
- SDcard used		| EMMC

#### Additional Information (if applicable):
- Software title	| Jacket
- Was the software title installed freshly or updated/migrated? existing
- Can this issue be replicated on a fresh installation of DietPi? don't know
- `dietpi-bugreport` ID

#### Steps to reproduce:
1. ... Update to 6.18
2. ... Jackett doesn't launch

#### Expected behaviour:
Jackett launch

#### Actual behaviour:
```
Dec  5 22:15:38 DietPi systemd[1]: Started Jackett Daemon.
Dec  5 22:15:40 DietPi mono[1426]: [Warn] ParsingService: Unable to parse movie info from path /mnt/ExtremeSSD/transmission/scream-queen(rutracker.org).avi
Dec  5 22:15:41 DietPi mono[2374]: 12-05 22:15:41 Info Starting Jackett v0.10.504.0
Dec  5 22:15:43 DietPi mono[2374]: 12-05 22:15:43 Error System.BadImageFormatException: Could not resolve field token 0x04000006, due to: Could not load type of field 'Microsoft.AspNetCore.WebHost+<>c:<>9__8_0' (2) due to: Could not load file or assembly 'Microsoft.AspNetCore.Server.Kestrel.Core, Version=2.1.3.0, Culture=neutral, PublicKeyToken=adb9793829ddae60' or one of its dependencies. assembly:/opt/jackett/Microsoft.AspNetCore.dll type:<>c member:(null)
Dec  5 22:15:43 DietPi mono[2374]: File name: 'Microsoft.AspNetCore'
Dec  5 22:15:43 DietPi mono[2374]:   at Jackett.Server.Program.CreateWebHostBuilder (System.String[] args, System.String[] urls, System.String contentRoot) [0x00000] in <f973eff5f36f4333aa34fefc0b6cccca>:0
Dec  5 22:15:43 DietPi mono[2374]:   at Jackett.Server.Program.Main (System.String[] args) [0x002b7] in <f973eff5f36f4333aa34fefc0b6cccca>:0  System.BadImageFormatException: Could not resolve field token 0x04000006, due to: Could not load type of field 'Microsoft.AspNetCore.WebHost+<>c:<>9__8_0' (2) due to: Could not load file or assembly 'Microsoft.AspNetCore.Server.Kestrel.Core, Version=2.1.3.0, Culture=neutral, PublicKeyToken=adb9793829ddae60' or one of its dependencies. assembly:/opt/jackett/Microsoft.AspNetCore.dll type:<>c member:(null)
Dec  5 22:15:43 DietPi mono[2374]: File name: 'Microsoft.AspNetCore'
Dec  5 22:15:43 DietPi mono[2374]:   at Jackett.Server.Program.CreateWebHostBuilder (System.String[] args, System.String[] urls, System.String contentRoot) [0x00000] in <f973eff5f36f4333aa34fefc0b6cccca>:0
Dec  5 22:15:43 DietPi mono[2374]:   at Jackett.Server.Program.Main (System.String[] args) [0x002b7] in <f973eff5f36f4333aa34fefc0b6cccca>:0
Dec  5 22:15:43 DietPi mono[2374]: Unhandled Exception:
Dec  5 22:15:43 DietPi mono[2374]: System.BadImageFormatException: Could not resolve field token 0x04000006, due to: Could not load type of field 'Microsoft.AspNetCore.WebHost+<>c:<>9__8_0' (2) due to: Could not load file or assembly 'Microsoft.AspNetCore.Server.Kestrel.Core, Version=2.1.3.0, Culture=neutral, PublicKeyToken=adb9793829ddae60' or one of its dependencies. assembly:/opt/jackett/Microsoft.AspNetCore.dll type:<>c member:(null)
Dec  5 22:15:43 DietPi mono[2374]: File name: 'Microsoft.AspNetCore'
Dec  5 22:15:43 DietPi mono[2374]:   at Jackett.Server.Program.CreateWebHostBuilder (System.String[] args, System.String[] urls, System.String contentRoot) [0x00000] in <f973eff5f36f4333aa34fefc0b6cccca>:0
```

#### Extra details:

Fourdee, 2018-12-06:
@19eighties 

Many thanks for the report 👍 

We'll try to replicate.

Fourdee, 2018-12-06:
@19eighties 

Unable to replicate in local testing. Please can you send us a bug report:
```
dietpi-bugreport
```

19eighties, 2018-12-07:
Thanks for looking into it. 

Just sent: Bug report sent, reference code: 08f012d0-1e7f-41f7-af5e-d7d1a1701913

Fourdee, 2018-12-17:
@19eighties 

Apologies for the delay.

```
[  304.188961] EXT4-fs (mmcblk0p7): error count since last fsck: 1
[  304.188995] EXT4-fs (mmcblk0p7): initial error at time 1533094581: ext4_find_entry:1450: inode 393400
[  304.189010] EXT4-fs (mmcblk0p7): last error at time 1533094581: ext4_find_entry:1450: inode 393400
[86807.851355] EXT4-fs (mmcblk0p7): error count since last fsck: 1
[86807.851401] EXT4-fs (mmcblk0p7): initial error at time 1533094581: ext4_find_entry:1450: inode 393400
[86807.851412] EXT4-fs (mmcblk0p7): last error at time 1533094581: ext4_find_entry:1450: inode 393400
```

Appears filesystem issue has occurred, may be the underlying cause of the issue.

Please can you verify PSU and SD/EMMC used?

---

```
● jackett.service - Jackett Daemon
   Loaded: loaded (/etc/systemd/system/jackett.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-12-07 11:57:20 PST; 1s ago
 Main PID: 23030 (mono)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/jackett.service
           └─23030 /usr/bin/mono /opt/jackett/JackettConsole.exe

Dec 07 11:57:20 DietPi systemd[1]: Started Jackett Daemon.
```

Service running fine. Although mono errors in log. 

---

@19eighties 
Could be a bug in Jackett itself, please try reinstalling the application:
```
dietpi-software uninstall 147
dietpi-software install 147
```

19eighties, 2018-12-27:
I did a reinstall and now Jackett works again.

closed 2018-12-27 by 19eighties