MichaIng/DietPi #2331: Asus TB | PS3 Eye camera not working

adamotte, 2018-12-06:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.19.6`
- Distro version	| `9.6`
- Kernel version	| `Linux DietHA 4.4.71+`
- SBC device		| `Asus Tinker Board (armv7l)`
- Power supply used	| `Ankru 5V 3A`
- SDcard used		| `No` 
- eMMC                   | `Yes`

#### Additional Information (if applicable):
- Software title	| MotionEye
- Was the software title installed freshly or updated/migrated? | `freshly and updated`
- Can this issue be replicated on a fresh installation of DietPi? |`Yes`

#### Steps to reproduce:
1. Plug in any USB ports the PS3 Eye camera
2. Run `lsusb`
3. Run `v4l2-ctl --list-devices`

#### Expected behaviour:
Something like : 
```tty
root@DietHA:~# lsusb
Bus 001 Device 001: ID 1415:2000 Nam Tai E&E Products Ltd. or OmniVision Technologies, Inc. S
[...]

root@DietHA:~# v4l2-ctl --list-devices
USB Camera-B3.04.06.1 (usb-bcm2708_usb-1.2.4.1) :
/dev/video0
```

#### Actual behaviour:
```tty
root@DietHA:~# lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 1415:2000 Nam Tai E&E Products Ltd. or OmniVision Technologies, Inc. S
Bus 001 Device 003: ID 0451:16a8 Texas Instruments, Inc.
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bda:481a Realtek Semiconductor Corp.
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@DietHA:~# v4l2-ctl --list-devices
Failed to open /dev/video0: No such file or directory
```

#### Extra details:
FYI, the audio part (micro) seems working because i'm using it with [snips.ai](https://snips.ai/)

MichaIng, 2019-03-12:
@adamotte 
Sorry for this was forgotten. Did you manage it in between?

If not please check if the devices actually exist:
`ls -Al /dev/video*`

MichaIng, 2019-05-26:
I close this due to missing user reply. Feel free to reopen if the issue persists.

closed 2019-05-26 by MichaIng

riker09, 2020-03-23:
I believe I'm having this issue as well. I'm not using DietPi, but Armbian. But it seems that the driver module `gspca_ov254` is missing from the kernel. I stumbled across this when I moved from kernel v4.4.x to v5.4.x and my PS3 Eye stopped working on my NanoPi M4.

I believe this can explain why I think the module is missing:

```bash
volker@nanopim4:~$ sudo modinfo gspca_main
filename:       /lib/modules/5.4.26-rockchip64/kernel/drivers/media/usb/gspca/gspca_main.ko
version:        2.14.0
license:        GPL
description:    GSPCA USB Camera Driver
author:         Jean-François Moine <http://moinejf.free.fr>
srcversion:     622BE485A3666FE37396CA5
depends:        videobuf2-v4l2,videodev,videobuf2-vmalloc,videobuf2-common
intree:         Y
name:           gspca_main
vermagic:       5.4.26-rockchip64 SMP preempt mod_unload aarch64
parm:           debug:1:probe 2:config 3:stream 4:frame 5:packet 6:usbi 7:usbo (int)

volker@nanopim4:~$ sudo modinfo gspca_ov254
modinfo: ERROR: Module gspca_ov254 not found.
```

I'm not a kernel guru and already spent quite some time finding this. I will try and compile that module myself and see if that does something.

MichaIng, 2020-03-23:
@riker09 
Yes, probably this driver can be enabled/disabled in kernel config and has been disabled for some reason by Armbian. Compiling a kernel module would be easiest solution that can be adopted for others as well, without the need to maintain an own kernel branch. This issue seems then somehow similar as #3435.

Generally I do not see us compiling own kernels, but I could imagine us hosting some module compiling + integration scripts, via DKMS or simple /etc/kernel/postinst.d/ scripts. As long as kernel headers are available, this is usually painless and does not require much maintenance.