MichaIng/DietPi #2332: Images | Drop Jessie images

MichaIng, 2018-12-06:
Since all remaining Stretch related bugs are solved (mainly Odroid C2 Kodi HW acceleration + desktop Xorg start), we should take out all Jessie images from official download source. As well we should disable support in DietPi-PREP, only allowing to upgrate Jessie images to Stretch.

However they could be kept in e.g. legacy sub dir, but then with a README.txt that they are not recommended and support might be dropped.

We should then keep an eye on existing Jessie install count (survey) and encourage users to switch to a Stretch image, e.g. via notification in banner and/or on next update.

I suggest we either at timestamps to survey report or simply remove older uploads to have more current numbers.
Currently there are 1-2% Jessie installs.

Waiting for e.g. 3 months, checking Stretch vs Jessie counts on meanwhile uploads, then create final Jessie update and drop support from code.

Feedback suggestions on this are highly appreciated. We will definitely drop Jessie support by times, to free special handling on code and reduce testing effort, especially when it comes to official Buster release, expected in summer 2019.
But we want to make this change as smooth as possible for Jessie users.

Atavic, 2018-12-09:
They should be kept for some time.

MichaIng, 2018-12-09:
@Atavic 
Do you have a reason for that? Is there anything that works better on a certain Jessie setup, or works on Jessie only?

Of course DietPi code support for Jessie should stay longer, but the question is, if there is any reason to still ship Jessie on fresh images, which would further delay the possibility to move on with the code as well.

Atavic, 2018-12-09:
No reasons except the doubt that you clarified:

> Of course DietPi code support for Jessie should stay longer

Debian has this page: https://www.debian.org/releases/jessie/debian-installer/

So it could be fine to have it still available for a while, even if superseded by following versions.

Fourdee, 2018-12-10:
@MichaIng 

> For now it's just about preventing new users from installing a fresh Jessie DietPi image.

Yep agree, no benefit to Jessie over Stretch. And it increases our level of support effort which should be spent elsewhere (eg: buster).

My preference:
- Drop all Jessie images
- During update, if Jessie, update to latest version at time we drop (eg: v6.20), then prevent further updates (allows us to remove Jessie code in future updates)
- Prompt users with existing installations that during update, Jessie support is no longer available. System will still function, but no future updates from our end.

Atavic, 2018-12-10:
> During update, if Jessie, update to latest version

Isn't the prompt warning enough?

MichaIng, 2018-12-10:
@Atavic 
When it's about removing Jessie support from code, a final update is required to prevent further updates of Jessie systems to unsupported code.

We did something similar when dropping wheezy support: https://github.com/Fourdee/DietPi/tree/wheezy-final-v120
- A last update was done to final wheezy branch, further updates from there blocked with information that a new image is required.

MichaIng, 2018-12-11:
One issue:
- For Odroid C1 we only have a Jessie image, because also Meveric offers only a Jessie image: https://forum.odroid.com/viewforum.php?f=114
- About mali drivers and Kodi support on manually distr-upgraded Meveric Jessie image: https://forum.odroid.com/viewtopic.php?f=114&t=30053
  - It could be tested, if new required packages are in Meverics repo, but I doubt.

MichaIng, 2019-01-26:
@Fourdee 
I think we should upload an Odroid C1 Stretch image. Meveric does not offer one, but the related forum thread includes users reporting successful distro upgrade. With DietPi-PREP we have best chances that this will work, however the system should be tested, perhaps added as experimental/beta state first.

MichaIng, 2019-03-25:
amd64 packages from the Debian main repo are not available anymore, similar to what happened to the security repo some time ago: https://github.com/MichaIng/DietPi/issues/2665

MichaIng, 2019-04-27:
Issues with Jessie are growing, leading to testing and workaround effort that we can not handle much longer, e.g.: https://github.com/MichaIng/DietPi/issues/2665

We will support Jessie for at least one version longer (until v6.24, at least) but move systems to the dedicated `jessie-support` branch already with v6.23. It is better to start with this earlier than later since we don't know how long users wait with the update. This should also raise attention and be a warning/push for Jessie users to migrate to Stretch as soon as they can.

Jessie systems are less than _0.5%_, constantly decreasing: https://dietpi.com/survey/#distro

Fourdee, 2019-04-29:
C1:
- Image fails to update correctly as jessie-backport fails:
- Reach DietPi-Software, unable to continue, stuck in loop.

@MichaIng 

I'll update the image as a final image with new jessie branch.

MichaIng, 2019-04-29:
@Fourdee
Jep jessie-backports and jessie-updates branches do not exist anymore in Debian repo. I already added a pre-patch and DietPi-PREP fix for this. New image required since the existing one does not load pre-patches yet.

MichaIng, 2019-06-25:
Okay, v6.25 is the last version with Jessie support. `Jessie-support` branch has been created already and systems moved to this branch since v6.23 automatically.

v6.26 will have all Jessie related code removed completely. The Jessie-support branch will stay, most likely until Stretch support is about to be dropped (~2 years from now) and may receive very important bug fixes, when reported or some slight Jessie tailor when someone is in mood to do so.

However again I recommend to switch to Stretch, where possible, also for security reasons.

MichaIng, 2019-07-06:
PR up for dropping Jessie support and related code cleanup: https://github.com/MichaIng/DietPi/pull/2963

closed 2019-08-11 by MichaIng