MichaIng/DietPi #2342: etherpad-lite

msongz, 2018-12-12:
## Creating a software request:

#### Give us some formal software information:
- Software title |  etherpad-lite
- Short description |  Really real-time collaborative document editing
- GitHub URL (if available) | https://github.com/ether/etherpad-lite
- External URL (if available) | http://etherpad.org/
- (Official) install documentation | 
https://github.com/ether/etherpad-lite/blob/master/README.md
#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

Atavic, 2018-12-13:
> Requirements preferred: nodejs >= 8.9

Actually it has no alternatives, except for a full LAMP Stack.

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+23