MichaIng/DietPi #2345: Eee E ee Eeee

mrmajik45, 2018-12-12:
Let E be Free. [E](https://github.com/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee)
(This is a protest for the E repository as GitHub has taken it down.)
Just posting this where I can.

mrmajik45, 2018-12-12:
[Post about being closed](https://linuxwit.ch/blog/2018/12/everything-that-lives-is-designed-to-end/)
[A repo name longer than ten characters](https://github.com/rust-lang-nursery/edition-guide)

Fourdee, 2018-12-12:
@mrmajik45 

Thanks for the heads up 👍. Interesting 😄

closed 2018-12-12 by Fourdee

MichaIng, 2018-12-12:
@mrmajik45 
Jep interesting. I mean, regardless of the purpose of this repo, indeed it is unreadable and visually ugly to have orga and repo both named just with a bunch of `e`'s, so much, that you carefully need to count (or copy&paste) to clone and such. Looks like spam on first view, at least to me, not knowing the repo 😉. So I understand the purpose of GitHub, forcing the owner to limited the amount of `e`'s to a smaller, more readable and reasonable number. Of course this is a lower max amount then for other orgas/repo where it is not just a single repeating character, but a meaningful name. There is no rule about max characters, AFAIK, it is about the meaningless (?) high amount of a single char.

BUT: Veeery bad practice to simply block access without informing and talking to the owner BEFOREHAND and find a solution together, since this is a single case issue. 😠

Evil voices might now associate this with Microsoft eating GitHub 😆.