MichaIng/DietPi #2346: Upgrade to .17 wipes wifi settings on RPi 3

GrahamGIGA, 2018-12-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -a`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

GrahamGIGA, 2018-12-12:
Upgrading via dietpi-update appears to wipe existing wi-fi settings out.

Using the network configurator shows the (correct) entered settings for SSID and Password, but the default IP is assigned, despite DietPi claiming to be connected to the corrct SSID.

Mounting the card on MacOS, this is the dietpi.txt file that is extant (after the upgrade and after setting the SSID and password via the configurator:

"#Modifications to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

#NB: This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
#NB: Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# DietPi-Automation
# All settings below will be applied on the 1st boot of DietPi, ONCE.
#------------------------------------------------------------------------------------------------------

##### Networking Options #####

#	If both Ethernet and Wifi are enabled, Wifi will take priority and Ethernet will be disabled.
#		1=enabled
AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=0

#	Enter your Wifi details below, if applicable (Case Sensitive).
AUTO_SETUP_NET_WIFI_SSID=MySuperDooperWiFi
AUTO_SETUP_NET_WIFI_KEY=0123456789
#	available | WPA-PSK / WPA-EAP / NONE | WEP=No longer supported
AUTO_SETUP_NET_WIFI_KEYMGR=WPA-PSK

#  For WPA2 Enterprise uncomment the following entries and set accordingly
#AUTO_SETUP_NET_WIFI_PROTO=RSN
#AUTO_SETUP_NET_WIFI_PAIRWISE=CCMP
#AUTO_SETUP_NET_WIFI_AUTH_ALG=OPEN
#AUTO_SETUP_NET_WIFI_EAP=PEAP
#AUTO_SETUP_NET_WIFI_IDENTITY=<userid>
#AUTO_SETUP_NET_WIFI_PASSWORD=<userpassword>
#
# If your WPA2 Enterprise requries a certificate copy the cert to this /boot directory and enter the name on the following line
#AUTO_SETUP_NET_WIFI_CERT=wpa2ent_cert.cer
# Uncomment either or both lines below and set accordingly for your WPA2 Enterprise needs.
#AUTO_SETUP_NET_WIFI_PHASE1=
#AUTO_SETUP_NET_WIFI_PHASE2=auth=MSCHAPV2

#	Enter your Static Network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.0.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=8.8.8.8

#	Hostname
AUTO_SETUP_NET_HOSTNAME=DietPi

#	Force ethernet speeds
#		NB: This is mainly aimed towards Pine A64's which may have a HW issue that causes unstable 1Gbit link.
#		0=automatic speed | 10 = 10mbit, 100 = 100mbit etc
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

##### Misc Options #####

#	Size of swapfile to generate (MB)
#		0=Disabled | 1=auto (2GB-RAM = size) | 2+=manual
AUTO_SETUP_SWAPFILE_SIZE=1072
#		Optional swapfile location
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap

# Unmask (enable) systemd-logind service, which is masked by default on DietPi 
AUTO_UNMASK_LOGIND=0

[... 165 lines cut ...]
SOFTWARE_WIFI_HOTSPOT_SSID=DietPi-HotSpot
#	minimum of 8 characters
SOFTWARE_WIFI_HOTSPOT_KEY=dietpihotspot
SOFTWARE_WIFI_HOTSPOT_CHANNEL=3

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=Fourdee
CONFIG_CPU_DISABLE_TURBO=0
CONFIG_ENABLE_IPV6=1
CONFIG_PREFER_IPV4=1
SOFTWARE_XORG_DPI=96
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://google.com"

Changing the dietpi.txt file to the correct settings does NOT result in correction of the issue.

[image]
[image]
[image]
[image]

Fourdee, 2018-12-13:
@GrahamGIGA 

Hi Graham,

Many thanks for the report 👍 

In ```dietpi-config``` under WiFi options, please can you toggle:
- ```Change Mode: DHCP```
- ```Save changes```

Once completed, WiFi should now be connected to network, at which point please can you send me a bugreport:
```
dietpi-bugreport
```

Also, can you remember which version of DietPi you were running, prior to the update?

---

The ```dietpi.txt``` options are only used on 1st run, after which point ```dietpi-config``` must be used to make any changes.

Very strange this has switched to STATIC IP during the update. I can only suspect ```$G_DIETPI_INSTALL_STAGE``` was not updated to 1 during your 1st run. Regardless, i'll do some local testing and see if we can replicate.

MichaIng, 2018-12-13:
@GrahamGIGA 
Thanks for your detailed report.

Note that we changed where to set WiFi settings to dietpi-wifi.txt on first boot. I have to check whether we correctly port existing settings from dietpi.txt on update.

If you set WiFi settings afterwards and you don't choose DHCP, but static IP, use the "copy" option to copy currently applied IP and such into static settings. Usually DHCP is what you want (first) to get correct IP range, mask, gateway and DNS server from AP (router), then copy the applied settings to static IP, if you don't want the DHCP server from then on.

GrahamGIGA, 2018-12-13:
Hi Dan,

I had missed the change to static - thanks.

There is something very odd going on here.  I now have the Pi working again, with an assigned IP of .252, which is fine EXCEPT that the DHCP server [should] allocates a predefined IP based on the device.  The setting in the server appeared to be correct, but was ignored.  

I'll run the pi as is for a couple of days.

You have bug report:  Bug report sent, reference code: a1fceb05-eda7-4eaf-9b99-38e5d4cfc4ad

To your question of prior version, I am unsure but THINK it was .14. I have two that updated fine (from .16) and two that have not - they are both older versions.

I do find that the wifi setup and settings (UI) are poorer than they used to be and it is much easier to make a change that is not intended.

One final thing: during the changing of wireless ettings, after one reboot the system insisted that I must Sudo dietpi-config.  Only the once, the next time it worked as expected, i.e. from the command line.

Cheers

GrahamGIGA, 2018-12-13:
One final, final (?) comment on wireless.  On the 3B there is an on card wireless and I have a Rosewill USB dongle plugged in.  This is for reliability, the onboard was (is?) poor in earleir versions.  When I diasble onboard wifi what am I disabling - the built in or all?

The dongle isn't distnguished in the setup menu.

Cheers

Fourdee, 2018-12-13:
@GrahamGIGA 

Hi,

>  IP of .252, which is fine EXCEPT that the DHCP server [should] allocates a predefined IP based on the device

You'll need to verify the MAC address of the USB WiFi device matches .252 in the router control panel. Failing that, i'am unsure as router setup varies.

> One final thing: during the changing of wireless ettings, after one reboot the system insisted that I must Sudo dietpi-config.

Sudo is only required if logged in with the dietpi user ```dietpi```.  If you login as root, ```sudo``` is not required. 

> This is for reliability, the onboard was (is?) poor in earleir versions. 

Onboard WiFi on the RPi is not as good as some USB dongles, however, as long as the range is acceptable, onboard WiFi is stable.

When you disable onboard WiFi in ```dietpi-config```, this will disable the onboard RPi brcm module from loading. Essentially preventing it from being available.
Once disabled, you will need to re-setup WiFi in ```dietpi-config``` as the USB wlan adapter will change to wlan0.

I believe the main here is confusion in the ```dietpi-config``` UI for WiFi network setup, as I'am unable to see any bugs our end at the moment.

GrahamGIGA, 2018-12-14:
Hi Dan,

Well the system definitely changed the DHCP setting to "static" AND headless during the update :-)

Fourdee, 2018-12-15:
@GrahamGIGA 

Is this the same system, PSU and SD card as: https://github.com/Fourdee/DietPi/issues/2353?

If not, please can you list the above for this system also?

GrahamGIGA, 2018-12-15:
No, this is a differemt system.  Now working fine after the "fix" to the static IP and headless settings.

Fourdee, 2019-01-28:
@GrahamGIGA 

Unable to replicate this in our local testing. 

I believe this is now resolved? I'll mark as closed. If problems persist, please reopen.

closed 2019-01-28 by Fourdee