MichaIng/DietPi #2348: RockPro64 updates automatically on 1st boot, reboots on his own, does not start anymore

1uc4, 2018-12-12:
## Creating a bug report/issue:
RockPro64 updates automatically on 1st boot, reboots on his own, does not start anymore

#### Required Information:
- DietPi version	| v6.14 --> v6.19
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -a`
- SBC device		| RockPro64
- Power supply used	| ROCKPro64 12V 3A EU POWER SUPPLY (from Pine store)
- SDcard used		| 32GB CLASS10 MICROSD CARD (from Pine store)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:
1. Flashed latest v6.14 for the RockProd64, default config
2. Booting with ethernet and HDMI, dietpi upgrade itselfs to v6.19 then reboots, no user interaction
3. Nothing happens, i'm left with a black screen with no signal
4. Plug - Unplug power to force reboot, still nothing, black screen, does not boot.

#### Expected behaviour:
usable system

#### Actual behaviour:
does not boot, black screen

#### Extra details:
I did not touch any dietpi.txt config, just flashed and booted

MichaIng, 2018-12-13:
@1uc4 
Thanks for your report.

I hope Fourdee finds some time to replicate, since it don't have a RockPro64 here.

Fourdee, 2018-12-13:
@1uc4 

Many thanks for the report 👍 

I suspect APT package kernel/uboot update causing this, will verify with local testing.

Fourdee, 2018-12-13:
Notes:
- No kernel/uboot packages updated. Only standard Debian packages.
- Working fine after reboot.

Unable to replicate on my board with EMMC or SD. Regardless, will update the image based on Ayufan's latest:
- 0.7.9 our current image
- 0.7.11 Ayufan latest

https://github.com/ayufan-rock64/linux-build/releases/download/0.7.11/stretch-minimal-rockpro64-0.7.11-1075-arm64.img.xz

1uc4, 2018-12-13:
Thanks for looking into it. 
I will try plugging in the serial console tonight (will be my 1st time) to see if I can grub some additional output and attach it to this thread.
Some additional info:
- Board is v2.1, 4GB, 
- Boot form SD card, no eMMC module present on the board

Fourdee, 2018-12-13:
@1uc4 

Serial debug would be ideal.

In the mean time, i've updated our image based on Ayufan 0.7.11:
https://dietpi.com/downloads/images/DietPi_RockPro64-ARMv8-Stretch.7z

Please let us know if this resolves the issue.

1uc4, 2018-12-13:
It does. :1st_place_medal: 

I repeated the exact same steps, now on 1st boot it upgrades from v6.18 --> v6.19.7.
On reboot it does not come up, no white led turns up. Power button and Reset button don't help. An unplug-plug of the power connector with a pause of about 15-20sec in-between does the trick.

I will perform the procedure again on my second board and then try to see if I can capture serial debug on a soft reboot, which is failing.

Thanks again! :+1:

1uc4, 2018-12-13:
Here's the serial console output after issuing a reboot command, which fails to boot:
```
DDR Version 1.13 20180801
In
soft reset
SRX
Channel 0: LPDDR4,50MHz
CS = 0
MR0=0x98
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x4D
MR14=0x4D
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x4D
MR14=0x4D
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
Channel 1: LPDDR4,50MHz
CS = 0
MR0=0x98
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x4D
MR14=0x4D
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x4D
MR14=0x4D
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
256B stride
channel 0
CS = 0
MR0=0x98
MR4=0x81
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x98
MR4=0x81
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 0 training pass!
channel 1 training pass!
change freq to 400MHz 0,1
channel 0
CS = 0
MR0=0x98
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x98
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 0 training pass!
channel 1 training pass!
change freq to 800MHz 1,0
ch 0 ddrconfig = 0x101, ddrsize = 0x2020
ch 1 ddrconfig = 0x101, ddrsize = 0x2020
pmugrf_os_reg[2] = 0x3AA1FAA1, stride = 0xD
OUT
U-Boot SPL board init

U-Boot SPL 2017.09-rockchip-ayufan-1033-gdf02018479 (Aug 06 2018 - 22:29:15)
booted from SD
Trying to boot from MMC2
mmc_load_image_raw_sector: mmc block read error
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```

Fourdee, 2018-12-16:
@1uc4 

> mmc_load_image_raw_sector: mmc block read error
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

Thanks for the serial debug 👍 

Yep, appears filesystem issue. Could be kernel related (latest image may resolve mmc/sd instability).

I also have a v2.1 rev board, was unable to replicate the issue you experienced. Very strange. I suspect kernel/stability issue during filesystem resize.

As the updated image resolved the issue, i'll mark this as closed. However, please reopen if required.

closed 2018-12-16 by Fourdee