MichaIng/DietPi #2352: Amiberry Fails to boot

symbios24, 2018-12-14:
Hello
I installed amiberry on the latest DietPi 6.19 in a Raspberry Pi3B installation ask me to set it to auto-start so i did.but after installation has finished amiberry is not starting
by doing a systemctl i see that the Amiberry service failed to start
I set in the dietpi-config in the auto-start section the normal boot for amiberry and i reboot
now i see in the Processes that dietpi diplays after has finished booting: 

INFO: SDL could not initialize SDL_ Error: No available video device
Internal error; file src/osdep/amiberry_gfx.cpp, line 324
/mnt/dietpi_userdata/amiberry/run.sh: line 3:  1183 aborted  ./amibbery-rpi3-sdl2-dispmanx -f conf/autostart.uae

FAILED: DietPi Process_Tool Amiberry: Affinity 0-3
FAILED: DietPi Process_Tool: An Issue Has Occurred

Any Suggestions?

Fourdee, 2018-12-14:
@symbios24 

Unable to replicate on fresh install and local testing.

Please can you check:
- ```dietpi-config``` > display options > GPU mem split is 128MB or higher
- ```dietpi-config``` > display options > Resolution > Ensure OpenGL is not being used

If problems persist, please send me a bug report and i'll check system configuration:
```
dietpi-bugreport
```

symbios24, 2018-12-15:
i have just send you the bug report.

symbios24, 2018-12-15:
the problem happens if i install Amiberry from Dietpi it self if i use the pre-made Dietpi amiberry image the problem is not there.
so the bug is in the DietPi Image

Fourdee, 2018-12-15:
@symbios24 

Very strange, I duplicated your system setup and was unable to replicate the issue. This was also using the latest DietPi standard image + installing AmiBerry on-top.

GPU mem check outs, all packages installed.

Hmmm, the only difference is you have a USB device attached. However, unless there is a power/hardware related issue, this should not have any effect.

As i'am unable to replicate in our testing, I'd need to have remote access to this system to debug this further. If this is something you would be willing to do, please: 
- enable port forward port ```22``` to the IP address of your RPi, on your router
- email daniel.knight@dietpi.com with the IP address of system

symbios24, 2018-12-15:
i dont thing it has to do with the usb keyboard/mouse i have connected or with the genuine rpi power adapter since it works fine with the pre-made amiberry image but not with the stock dietpi image,it just does not start the amiberry service at all.

symbios24, 2018-12-16:
its not possible to open a port since i`m taking internet from a neighbor with wifi.

symbios24, 2018-12-16:
since the pre-made amiberry image is old eg dietpi 6.6 this has to do something with it.for example something other is pre-installed there that makes work

symbios24, 2018-12-16:
since the pre-made amiberry image is old eg dietpi 6.6 this has to do something with it.for example something other is pre-installed there that makes work

How can i Install an older amiberry version 2.18.1 with sdl1 ??

Thanks

Fourdee, 2018-12-17:
@symbios24 

> How can i Install an older amiberry version 2.18.1 with sdl1 ??

We only provide the SDL2 version currently.

```DietPi_v6.6_RPi-ARMv6-Stretch_AmiBerry.img``` is actually an automated install image. AmiBerry is installed in exactly the same was as stock DietPi image, once it updates to latest version during 1st run.

> its not possible to open a port since i`m taking internet from a neighbor with wifi.

Ok, no worries, however, not much we can do our end to assist with debug/resolution.

symbios24, 2018-12-19:
Can you also add to the chromium browser installation the rpi-chromium-mods package (flash plugin and adblocker) i see that is missing.

Thanks

MichaIng, 2018-12-19:
@symbios24 
Should be possible. Just take care that these mods do not install some RPi-desktop UI related packages as dependencies, which would not fit do our non-RPi desktop.

symbios24, 2018-12-20:
i have tried it i works fine,it also plays 1080p videos from youtube smooth
Can you please release an updated image for Amiberry ??

Thanks

symbios24, 2018-12-20:
i have tried it i works fine,it also plays 1080p videos from youtube smooth
Is Sickrage removed ? i dont see it in the list to install it.
Can you please release an updated image for Amiberry ??

Thanks

symbios24, 2018-12-20:
It does not install anything else its safe,
i have tried it i works fine,it also plays 1080p videos from youtube smooth
Is Sickrage removed ? i dont see it in the list to install it.
Can you please release an updated image for Amiberry ??

Thanks

midwan, 2019-01-09:
@symbios24 
The fastest version and the one with the most features currently, is the SDL2 + Dispmanx for the RPI. The SDL1 is slightly slower and is missing the scaling options (Auto, Nearest Neighbour, Linear) which are provided by SDL2.

The error message you reported indicates that there's a problem with the SDL2 configuration, and it cannot find a supported video device. SDL2 can be configured to use different video devices, e.g. KMSDRM (recommended), X11, OpenGL, OpenGLES1, OpenGLES2, etc. Did you install anything else in your image that perhaps also used SDL2, and then changed it to Amiberry? Have you tried setting it up from scratch, installing DietPi and then Amiberry?

I also tried with the latest DietPi image, and installed Amiberry after that. It worked normally for me as well.

@Fourdee The only problem I noticed is this:
If you quit Amiberry, you return to the console which displays the message "Please login to continue". However, I cannot press enter to login on that console, it does not respond. I can use Ctrl-Alt-F1 to go to tty0 and login there of course, but this seems like a glitch that you might want to look at.

Fourdee, 2019-01-10:
@midwan 

> If you quit Amiberry, you return to the console which displays the message "Please login to continue". However, I cannot press enter to login on that console, it does not respond. I can use [...]

Yep, thanks. Did notice this the other day. Will take a look: https://github.com/Fourdee/DietPi/issues/2404

Fourdee, 2019-01-10:
Will mark this as closed as unable to replicate.