MichaIng/DietPi #2353: Version 6.12 dietpi-update fails

GrahamGIGA, 2018-12-14:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

root@Roon-Master-Bed:~# echo $G_DISTRO_NAME` or `cat /etc/debian_version
-bash: or: command not found
stretchcat /etc/debian_version
root@Roon-Master-Bed:~# 

- Kernel version	| `uname -a`

uname -a
Linux Roon-Master-Bed 4.14.61-v7+ #1132 SMP Tue Aug 7 15:36:25 BST 2018 armv7l GNU/Linux

- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)

root@Roon-Master-Bed:~# echo $G_HW_MODEL_DESCRIPTION
RPi 3 Model B (armv7l)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

Roon [but immaterail ??]

#### Steps to reproduce:

1. ...
[image]

2. ...
[image]

3.
Network seems fine:
[image]

#### Expected behaviour:

- ...
An update yo .17?  :-)

#### Actual behaviour:

- ...
See failure, above screenshot
#### Extra details:

- ...

root@Roon-Master-Bed:~# dietpi-bugreport
[  OK  ] Root access verified.
[  OK  ] DietPi-Drive_Manager | RootFS R/W access verified.

dpkg: error: unable to execute dpkg-query (dpkg-query): Input/output error
/DietPi/dietpi/dietpi-bugreport: line 131: /usr/bin/lsusb: Input/output error
/DietPi/dietpi/dietpi-bugreport: line 131: /sbin/blkid: Input/output error
/DietPi/dietpi/dietpi-bugreport: line 131: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
/DietPi/dietpi/dietpi-services: line 515: /usr/bin/cut: Input/output error
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[......] DietPi-Bugreport | (1/5) Testing connection to dietpi.com, please wait.[......] DietPi-Bugreport | (2/5) Testing connection to dietpi.com, please wait.[......] DietPi-Bugreport | (3/5) Testing connection to dietpi.com, please wait.[......] DietPi-Bugreport | (4/5) Testing connection to dietpi.com, please wait.[......] DietPi-Bugreport | (5/5) Testing connection to dietpi.com, please wait.[......] DietPi-Bugreport | Failed connection attempt (5/5), retrying.../DietPi/dietpi/func/dietpi-globals: line 1058: /usr/bin/tail: Input/output error
[FAILED] DietPi-Bugreport | Connection test: dietpi.com

[image]

louisefindlay23, 2018-12-15:
I have the same error.

Fourdee, 2018-12-15:
@GrahamGIGA @louisefindlay23 

> Input/output error

This indicates a possible hardware related issue with PSU/SD card. 
Please verify:
- Power supply used
- SD card used

Please can you both replicate the error, then paste results of:
```
dmesg
```

GrahamGIGA, 2018-12-15:
root@Roon-Master-Bed:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.61-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1132 SMP Tue Aug 7 15:36:25 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] On node 0 totalpages: 255488
[    0.000000] free_area_init_node: node 0, pgdat 80c85000, node_mem_map bd336000
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @be59d000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=27504eef-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 990960K/1021952K available (7168K kernel code, 575K rwdata, 2076K rodata, 1024K init, 706K bss, 22800K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
                   lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8fef4   ( 576 kB)
                      .bss : 0x80c96f34 - 0x80d478b4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25267 entries in 75 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000276] Console: colour dummy device 80x30
[    0.000814] console [tty1] enabled
[    0.000851] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000891] pid_max: default: 32768 minimum: 301
[    0.001220] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001254] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002214] Disabling memory control group subsystem
[    0.002309] CPU: Testing write buffer coherency: ok
[    0.002737] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003152] Setting up static identity map for 0x100000 - 0x10003c
[    0.003291] Hierarchical SRCU implementation.
[    0.003986] smp: Bringing up secondary CPUs ...
[    0.004791] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005629] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006452] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006558] smp: Brought up 1 node, 4 CPUs
[    0.006631] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006652] CPU: All CPU(s) started in HYP mode.
[    0.006670] CPU: Virtualization extensions available.
[    0.007576] devtmpfs: initialized
[    0.017816] random: get_random_u32 called from bucket_table_alloc+0xfc/0x24c with crng_init=0
[    0.018505] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.018748] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.018791] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.019366] pinctrl core: initialized pinctrl subsystem
[    0.020163] NET: Registered protocol family 16
[    0.022780] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.027570] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.027602] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 832 lines cut ...]
[255098.994328] Buffer I/O error on device mmcblk0p2, logical block 120324
[261801.054167] EXT4-fs warning (device mmcblk0p2): ext4_end_bio:323: I/O error 10 writing to inode 3294 (offset 0 size 0 starting block 132612)
[261801.054184] Buffer I/O error on device mmcblk0p2, logical block 120324
[... cut ...]
[299150.244239] Buffer I/O error on device mmcblk0p2, logical block 120325
[304983.291381] EXT4-fs warning (device mmcblk0p2): ext4_end_bio:323: I/O error 10 writing to inode 3294 (offset 0 size 0 starting block 132613)
[304983.291397] Buffer I/O error on device mmcblk0p2, logical block 120325
[311398.922712] EXT4-fs warning (device mmcblk0p2): ext4_end_bio:323: I/O error 10 writing to inode 3294 (offset 0 size 0 starting block 132613)
[311398.922728] Buffer I/O error on device mmcblk0p2, logical block 120325
root@Roon-Master-Bed:~#

Fourdee, 2018-12-16:
@GrahamGIGA 

> [  479.193320] mmc0: cmd op 18 arg 0xc7d7b8 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[  479.193328] mmc0: data blocks 150 blksz 200 - err 0
[  479.193338] mmc0: stop op 12 arg 0x0 flags 0x95 - resp 00000000 00000000 00000000 00000000, err 0
[  479.193343] mmc0: =========== REGISTER DUMP ===========
[  479.193348] mmc0: SDCMD  0x00004052
[  479.193353] mmc0: SDARG  0x00c7d7b8
[  479.193359] mmc0: SDTOUT 0x017d7840
[  479.193364] mmc0: SDCDIV 0x00000003
[  479.193369] mmc0: SDRSP0 0x00000900
[  479.193374] mmc0: SDRSP1 0x00001269
[  479.193379] mmc0: SDRSP2 0xffffffff
[  479.193384] mmc0: SDRSP3 0x0002c01f
[  479.193389] mmc0: SDHSTS 0x00000080
[  479.193394] mmc0: SDVDD  0x00000001
[  479.193400] mmc0: SDEDM  0x00010801
[  479.193405] mmc0: SDHCFG 0x0000040e
[  479.193410] mmc0: SDHBCT 0x00000200
[  479.193415] mmc0: SDHBLC 0x00000018
[  479.193420] mmc0: ===========================================
[  479.194780] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[  479.194884] mmcblk0: error -110 transferring data, sector 13096888, nr 336, cmd response 0x900, card status 0xb00
[  482.251459] mmc0: card never left busy state
[  482.251477] mmc0: tried to reset card, got error -110

Yep, we have a filesystem failure. 

I suspect PSU is the cause of this, we highly recommend using the official RPi PSU's only for stability. SD card may also be failing, however, a full format of the SD card is also recommended: https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267

You'll need to rewrite the DietPi image once PSU and SD card replaced/resolved. If problems persist after these are resolved, could indicate a hardware failure on the RPi itself.

Fourdee, 2018-12-16:
@GrahamGIGA 

> Sorry about the strikethrough, no idea why GH does that on a standard paste....

Add three tilde keys above and below the code you pasted.

louisefindlay23, 2018-12-16:
I get Write error - write (28: No space left on device) IO Error saving source cache The package lists or status file could not be parsed or opened. I used df -h and I have over 70% left and I still get it on a free reinstall

Fourdee, 2018-12-17:
@GrahamGIGA 

Ah, is this one of those 15v AMP HAT's that also powers the RPi? If so, could be a design flaw with the HAT, or, the PSU is not sufficient enough?

---

@louisefindlay23 

Very strange, would you be able to send us a bug report?
```
dietpi-bugreport
```

MichaIng, 2019-02-27:
Will mark as closed due to missing user reply and since most likely hardware related. Feel free to reopen, if required.

closed 2019-02-27 by MichaIng