MichaIng/DietPi #2359: DietPi-Software | Nextcloud "First Run Wizard" is missing/ doesn't initiate

minnux, 2018-12-20:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=7
G_GITBRANCH=master
G_GITOWNER=Fourdee
- Distro version	| stretch
- Kernel version	| 
Linux DietPi 4.4.154-1124-rockchip-ayufan-ged3ce4d15ec1 #1 SMP Mon Oct 22 20:59:41 UTC 2018 aarch64 GNU/Linux
- SBC device		| RockPro64 (aarch64)
- Power supply used	| 12V 5A US Type Power Supply for ROCKPro64 SBC
- SDcard used		| 16GB EMMC

#### Additional Information (if applicable):
- Software title	| Nextcloud 15
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes
-  078be993-a37f-4fdd-acf5-4dcd2c24ab27 ID

#### Steps to reproduce:

Install Nextcloud from dietpi-software, reboot and login to the web interface

#### Expected behaviour:

- First Run Wizard should initiate

#### Actual behaviour:

- Only fields for UN and PW, no wizard, also no link to First Run Wizard on personal page as app description suggests.

#### Extra details:

- New Nextcloud installation on rockpro64 from EMMC. Installed Nextcloud rom dietpi-software but there is no option to select where nextcloud data will be stored. Reading the Nextcloud docs there should be a first run wizard that pops up when going to your Nextcloud install's address for the first time, but all I'm greeted with is a login panel that uses admin and password. No wizard is present. Once I'm logged in there is no link to the wizard even though when I go to the apps section it says first install wizard is enabled, clicking on it gives me this message:

_This application enables a pop-up window when a user first logs into Nextcloud. This window welcomes new users to Nextcloud, and contains links to the standard desktop client, Android app, and the iOS app. The window also shows a link for accessing Nextcloud via WebDAV. At any time, the first run wizard button can be opened again via the link on the personal page. The First run wizard can be customized to meet specific design goals, or to change links and messages. Create an appropriate theme following the Theming documentation, and changes to the look and feel of this Wizard are simple. More information is available in the First run wizard documentation, as well as the theming documentation._

There is no first run wizard link on the personal page. All I need to do is select my mounted external drive to be the main nextcloud storage and I don't have that option. 

The install itself went smoothly with no perceptible errors. Installed several times with same results.

Fourdee, 2018-12-22:
@minnux 

Hi,

DietPi install of Nextcloud automates the initial setup with user + password as set during 1st run of system setup.

The default data directory is:
```
/mnt/dietpi_userdata/nextcloud_data
```

> There is no first run wizard link on the personal page. All I need to do is select my mounted external drive to be the main nextcloud storage and I don't have that option.

In DietPi we have a system called userdata (https://dietpi.com/phpbb/viewtopic.php?f=8&t=478). Nextcloud data directory can be moved by moving the DietPi userdata directory.

MichaIng, 2018-12-22:
@minnux 
That is the expected behaviour on our install.

The first run wizard is to setup database and admin user, but we already do this for you during our software config step. Simply log in via "admin" "dietpi", respectively the global password you chose on DietPi first run or dietpi.txt. Check out: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3026#p3026

This prevents you from setting up the database and allows us to do further setup of memory caching, file locking and other things.

> This application enables a pop-up window when a user first logs into Nextcloud.

What you mention there is not the first run setup wizard, but the login prompt. This however should appear also when installing via DietPi. Report back, if it is missing as well, but just tested locally and it appears here.

minnux, 2018-12-22:
Hello and thank you for the reply. These are the steps I have tried so far.

1. Install dietpi, format external drive via drivemanager, move user data to external drive via drivemanager, install Nextcloud.

Nextcloud errors out during install and fails to install.

2. Install dietpi, install Nextcloud, format external drive, move user data via drivemanager.

Nextcloud directory still shows EMMC being used as data directory when using web interface.

What should be my next option? I can provide exact error messages when I get home.

MichaIng, 2018-12-22:
@minnix
> Nextcloud directory still shows EMMC being used as data directory when using web interface.

Note that userdata is moved via symlink, so Nextcloud does not know that the actual location is on external drive. 

Many users did already report the issue to install Nextcloud when userdata was already moved. We were still not able to reproduce the issue. It was the MariaDB install that actually failed, right?

I will now change the install procedure so that MariaDB is first installed to the default directory and then moved+symlinked to userdata afterwards, which should resolve the issue.

MichaIng, 2018-12-23:
@minnux 
No worries, enjoy holidays 😃!

The info was already helpful, so default FS+partition table.
I will retry to format the whole drive as well, but AFAIK there is no difference, as long as you use a single partition anyway. Formatting whole drive will create a new partition table with one single partition.

On VM I used USB1 due to lack of VBox extension package, but will retry with it, just to be sure. Also I will switch tests to a real x86 notebook. I always forget that I have one left for testing 😆.

minnux, 2018-12-28:
Ok sorry for the delayed reply. I started over, reinstalled dietpi, and reformatted the HDD. I noticed that (I think) partition or whole drive option was missing now in the drive manager when I reformatted, and (I think) mariadb loads before php when installing Nextcloud. It was going by so fast that I didn't have time to take notice. Whatever changed though, it seems to have worked. Everything is up and running smoothly. The only problem I had was when trying to install Nextcloud Talk through dietpi-software, I got the error message that spreed could not install, but I was successfully able to install Talk through the Nextcloud web interface.

MichaIng, 2018-12-28:
@minnux 
I also recognized the issue with Nextcloud Talk. It is due to installing apps via `occ` command requires Redis to run and we start MariaDB only. Fixed with: https://github.com/Fourdee/DietPi/pull/2333/commits/d0f5827ca8e2e749eb0eb2feadc2f7b881f1f131

Great that it finally worked. How did you do it now, move userdata before or after Nextcloud install?

minnux, 2018-12-28:
I booted a fresh install of dietpi, went through the menus, but didn't install anything, then exited dietp-launcher at which point it says it will reboot . After reboot I used drive_manager to format the external drive to ext4 gpt, rebooted again, after startup I then moved userdata to external drive, tested data throughput, looked good so rebooted again (I voluntarily rebooted quite a bit just because I wanted no mistakes). Next I installed Nextcloud, rebooted, then certbot and configured through let's encrypt, rebooted, logged into nextcloud via web and everything was great. So far so good. Thanks again for the help.

MichaIng, 2018-12-28:
@minnux 
Okay great. It leaves the question open for me, why MariaDB in some cases fails to install with moved database. I am still thinking about switching the install procedure as mentioned above, but actually would love to find out the root cause.

Reboot after drive formt and moving userdata should be not required. However it does not hurt either and for sure is more failsafe 👍.

minnux, 2018-12-28:
Yes I figured I went overboard with the reboots but I was so tired of re-installing I didn't want any f ups this time lol.

MichaIng, 2019-02-27:
Mark as closed due to general cleanup. Question answered, this particular issue solved.

closed 2019-02-27 by MichaIng