MichaIng/DietPi #2360: Genius using the only archive format NOT supported by ESXi for the VMware image!

CollinChaffin, 2018-12-20:
Title sums it up - brilliant distro but boy would it be nice to only have to upload a mere 100mb to my ESXi like lightning and just quickly unzip, unrar, untar, etc.....but of course there is no native support for the one archive image selected for some reason for the VMware image - 7zip - on ESXi, and never has been.

Since I searched and did not see any existing request (very surprising) I'll put one in as it would really be nice not to have to either extract the 3+GIG to then re-compress it, or worse yet unzip it and actually wait to transfer the 3+GIG over SCP.  :)

Sorry for the slightly sarcastic tone but it really is frustrating when had it simply been a zip, rar, or tar it would be SO simple. :)

Fourdee, 2018-12-21:
@CollinChaffin 

Makes sense, we'll update the images to use .zip.

Fourdee, 2018-12-21:
Completed.

https://dietpi.com/download

closed 2018-12-21 by Fourdee